OPEnSLab-OSU/Loom-V4 #11: Relay missing digital control pin to specify

udellc, 2022-08-01:
**Describe the bug**
in the Max_Relay Example code.  Running the example with Max does not work.
Device shows "Connected". Serial printouts show receiving the correct state. There is nowhere for the user to specify which pin is the control pin for the Relay. This is done in hardware using the solder jumper, and expected to specify in hardware which solder jumper is used to control the relay from a digital pin.

**Hardware in Use**
Feather M0 WiFi, Power Relay Featherwing, Computer running Max

**To Reproduce**
Run the Max_Relay Example code and Max.

**Expected behavior**
Setting the state in Max should change the physical state of the relay. For now, it is only printing the correct state, but not changing setting the relay state in hardware via the digital pin.

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for Max controlled Relay</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Publishing</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Max</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Actuators</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Relay</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Relay</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">21</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_WIFI</font> <font color="#000000">wifi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">SECRET_SSID</font><font color="#434f54">,</font> <font color="#000000">SECRET_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_Max</font> <font color="#000000">maxMsp</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">,</font> <font color="#00979c">new</font> <font color="#000000">Loom_Relay</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Send and Recieve data from Max</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#000000">publish</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#d35400">subscribe</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">1000</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

**Additional context**
Add any other context about the problem here.

WL-Richards, 2022-08-01:
You can specify a control pin just pass in a number into the constructor of the Loom_Relay it defaults to pin 10

udellc, 2022-08-02:
Being able to specify the pin works, tested on Pin 13 LED.
But, declaring the add relay function as default like so: Loom_Relay() and also as Loom_Relay(10), still does not engage the relay. 
More testing needs done on the hardware to ensure pin 10 is going high and low. And that pin 10 is physicaly connected to control the relay (it should because it worked in Loom3).
Other check with the code to ensure all the pins are being handled in code correctly would be good.
[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: HIGH

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 12
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: LOW

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 13
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: HIGH

WL-Richards, 2022-08-02:
If it is triggering pin 13 correctly that definitely doesn't sound like a software thing

WL-Richards, 2022-08-02:
I would try again and see if it works now, closing for now, reopen if needed

closed 2022-08-02 by WL-Richards