OPEnSLab-OSU/Loom-V4 #6: Loom Max, Access Point WiFi Mode

udellc, 2022-08-01:
In Loom 4 Max, There are 2 ways to connect a device to the computer. 
Client Mode: Loom device WiFi connects to a WiFi router with an SSID and Password.
Access Point (AP) Mode: Loom WiFi device acts as the access point and the computer connects to it, no password required.

AP mode is used in classes a lot as many devices cannot be run on client from a central router due to too much traffic. 
AP Mode is a great start up state or default revert state if new router credentials need to be sent. 

Can we make an example of the Loom MPU to work in AP Mode? 
In Loom3 these were denoted as modes 7002 and 7003 or something like that. Arbitrary numbers to my understanding.

WL-Richards, 2022-08-01:
AP Mode is not yet implemented as it was causing problems in the old version this can be added but this should be logged under a todo in the Projects instead of an issue.

As for the 7002 and 7003 I'm 90% sure those are the ports it tries to talk over. Closing and adding an todo to the project

WL-Richards, 2022-08-07:
Added in most recent commit

closed 2022-08-08 by WL-Richards