OPEnSLab-OSU/Loom-V4 #8: Max Subscribe Example to set Neopixel

udellc, 2022-08-01:
Test the Max Subscribe feature to set the Neopixel color levels from Max.
Include Example in the Max examples on the board profile.

WL-Richards, 2022-08-01:
Wrriten and compiles

udellc, 2022-08-02:
Tested neopixel Max example and Neopixel acutator example, neither work.
Neopixel connected to Wattson A2 port.

udellc, 2022-08-02:
We'll need to test with hardware to sort this issues out I think. Hardware is in Gilmore Annex near WeatherChime equipment

udellc, 2022-08-04:
Tested on Hardware
Raw non-Max example works!
Max example does not work.
Neopixel connected to port A2
Need to test this together, or @WL-Richards can use hardware on desk in lab to work things out.

WL-Richards, 2022-08-04:
I think I may know why I can update it in a bit to test

WL-Richards, 2022-08-04:
I believe the newest release will have resolved this please test, closing until further notice

closed 2022-08-04 by WL-Richards