OPEnSLab-OSU/Loom-V4 #12: Prints for relay should reflect the pin number of the relay control pin

udellc, 2022-08-02:
**Describe the bug**
Serial prints of the Max_Relay example code report [Relay0] instead of the pin number of the specified control pin. this would provide useful feedback to the user to ensure they see which relay pin a status printout is referring to.

**Hardware in Use**
Power relay, Feather WiFi, Max on computer

**To Reproduce**
Run Max_Relay example with hardware and send commands from Max 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for Max controlled Relay</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Communication</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Max</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Actuators</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Relay</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Relay</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_WIFI</font> <font color="#000000">wifi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">SECRET_SSID</font><font color="#434f54">,</font> <font color="#000000">SECRET_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_Max</font> <font color="#000000">maxMsp</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">,</font> <font color="#000000">CommunicationMode</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">CLIENT</font><font color="#434f54">,</font> <font color="#00979c">new</font> <font color="#000000">Loom_Relay</font><font color="#000000">(</font><font color="#000000">10</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Send and Recieve data from Max</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#000000">publish</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#d35400">subscribe</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">1000</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

**Additional context**
Printout from the serial window:
[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: HIGH

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 12
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: LOW

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 13
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Relay0] Relay pin is set to: HIGH

WL-Richards, 2022-08-02:
It's not currently meant to print out the control pin which is why it prints out 0

WL-Richards, 2022-08-02:
Updated to reflect request

closed 2022-08-02 by WL-Richards