OPEnSLab-OSU/Loom-V4 #14: Stepper Motor Max example does not work

udellc, 2022-08-04:
**Describe the bug**
Stepper Motor Max example will connect to WiFi client mode; successfully sends packets. 
Max side prints errors from Convert_Incoming.js
See printouts and screenshots below

**Hardware in Use**
WiFi M0, Stepper motor shield, WiFi Router, computer running Max

**To Reproduce**
Run Max example Stepper

**Expected behavior**
Stepper sent packets should be properly parsed and are not on the Max side. Max parses well with other actuators and sensors, I suspect bad packets coming from stepper.
Messages from Max should control stepper, these are not indicated as received on the 

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for a Max controlled Stepper motor</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Communication</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Max</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Actuators</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Stepper</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Stepper</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_WIFI</font> <font color="#000000">wifi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">SECRET_SSID</font><font color="#434f54">,</font> <font color="#000000">SECRET_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_Max</font> <font color="#000000">maxMsp</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">,</font> <font color="#000000">CommunicationMode</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">CLIENT</font><font color="#434f54">,</font> <font color="#00979c">new</font> <font color="#000000">Loom_Stepper</font><font color="#000000">(</font><font color="#000000">0</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Send and Recieve data from Max</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#000000">publish</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#d35400">subscribe</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">1000</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>
**Arduino Serial prints**
[Manager] Initializing Modules...
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: LoomHub
[WiFi] Connected to network!
[WiFi] Verifying Connection to the Internet...
[WiFi] Ping Failed! Error Code: Unknown_Host
[WiFi] Successfully Initalized Wifi!
[WiFi] Device IP Address: 192.168.1.4
[WiFi] Device Subnet Address: 255.255.255.0
[Max Pub/Sub] Initializing Max Communication....
[Max Pub/Sub] Listening for UDP Packets on 9001
[Max Pub/Sub] Connections Opened!
[Max Pub/Sub] Initializing desired actuators...
[Max Pub/Sub] Successfully initialized actuators!
[Manager] ** Setup Complete ** 

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Max Pub/Sub] No message received!

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Max Pub/Sub] No message received!

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1,
    "ip": [
      192,
      168,
      1,
      4
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 3
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 192.168.1.255:8001
[Max Pub/Sub] Packet successfully sent!
[Max Pub/Sub] No message received!

**Max Screenshot**
Add any other context about the problem here.
[image]

WL-Richards, 2022-08-04:
Try the non-max example to see if it works

WL-Richards, 2022-08-04:
I'd also just try printing out the json received from the Connection node to see the JSON

WL-Richards, 2022-08-05:
Fixed, Motor Instance and Motor number were offset by

closed 2022-08-05 by WL-Richards