OPEnSLab-OSU/Loom-V4 #19: Weatherchimes WiFi example, removing/replacing coin cell causes strange behavior

udellc, 2022-08-25:
WeatherChimes WiFi example. Testing after programming and Normal behavior is verified. I want to remove the coin cell battery from the hypnos, and replace, so I can set the correct local time. 

I reveive printout that power was lost on the RTC, but no prompt to enter the time. Time proceeds in 2001.

These are the printouts I get:

[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Chime4.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[SHT31] Failed to initialize SHT31! Check connections and try again...
[TSL2591] Successfully initialized TSL2591!
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: OSU_Access
[WiFi] Connected to network!
[WiFi] Verifying Connection to the Internet...
[WiFi] Ping Failed! Error Code: Ping_TimeOut
[WiFi] Successfully Initalized Wifi!
[WiFi] Device IP Address: 10.248.203.184
[WiFi] Device Subnet Address: 255.254.0.0
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[Hypnos] Interrupt successfully attached!
[Hypnos] Current Time: 2000.01.01 00:01:34
[Hypnos] Next Interrupt Alarm Set For: 2001.08.28 01:28:14

Code here:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">4</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_2</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create the TSL2591 and SHT classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_SDI12</font> <font color="#000000">sdi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">11</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_WIFI</font> <font color="#000000">wifi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
[... 24 lines cut ...]

 &nbsp;<font color="#434f54">&#47;&#47; Set the RTC interrupt to trigger 10 seconds from when the device woke up</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">setInterruptDuration</font><font color="#000000">(</font><font color="#000000">TimeSpan</font><font color="#000000">(</font><font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure and package the data</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
[... cut ...]
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">reattachRTCInterrupt</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<font color="#434f54">&#47;&#47; Put the device into a deep sleep, operation HALTS here until the interrupt is triggered</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">sleep</font><font color="#000000">(</font><font color="#00979c">false</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

udellc, 2022-08-25:
Reprogramming the same sketch after this causes the device to hang. 
After reprogramming, these are the serial prints:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Chime4.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[SHT31] Successfully initialized SHT31!
[TSL2591] Successfully initialized TSL2591!
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: OSU_Access
[WiFi] Connected to network!
[WiFi] Verifying Connection to the Internet...
[WiFi] Successfully Pinged Google! Response Time: 10ms
[WiFi] Successfully Initalized Wifi!
[WiFi] Device IP Address: 10.248.203.184
[WiFi] Device Subnet Address: 255.254.0.0
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[Hypnos] Interrupt successfully attached!

hangs after this

udellc, 2022-08-25:
Trying to reset the RTC using the Hypnos_RTC_CustomTime example, I do not get prompted to input time over serial either and get strange time printouts:

[Hypnos] DS3231 Real-Time Clock Initialized Successfully!

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 3
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 4
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 5
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 6
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 7
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 8
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

udellc, 2022-08-25:
After reprogramming with the Hypnos_RTC_CustomTime example, removing and reinserting a new battery, I am getting prompted. Unclear if it is because reprogramming with the example and power cycling the RTC helped, or a new battery.

udellc, 2022-08-25:
However, after setting correct time and day in the user prompts, I still get strange time and date:
[Hypnos] RTC lost power, lets set the time!
[Hypnos] Please use your local time, not UTC!
[Hypnos] Enter the Year (Four digits, e.g. 2020)
[Hypnos] Year Entered: 2022
[Hypnos] Enter the Month (1 ~ 12)
[Hypnos] Month Entered: 8
[Hypnos] Enter the Day (1 ~ 31)
[Hypnos] Day Entered: 25
[Hypnos] Enter the Hour (0 ~ 23)
[Hypnos] Hour Entered: 10
[Hypnos] Enter the Minute (0 ~ 59)
[Hypnos] Minute Entered: 26
[Hypnos] Enter the Second (0 ~ 59)
[Hypnos] [Hypnos] Attempted to pull time when RTC was not previously initialized! Returned default datetime
Custom time successfully set to: 20:6.02.06 06:28:16
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 3
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 4
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 5
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 6
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 7
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

udellc, 2022-08-25:
Back to running Weatherchimes WiFi example and the correct time now appears to be in the RTC, but the set interrupt duration of 10 seconds sets the alarm to go off nor next year. See hypnos readout:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Chime5.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[SHT31] Successfully initialized SHT31!
[TSL2591] Successfully initialized TSL2591!
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: OSU_Access
[WiFi] Connected to network!
[WiFi] Verifying Connection to the Internet...
[WiFi] Successfully Pinged Google! Response Time: 10ms
[WiFi] Successfully Initalized Wifi!
[WiFi] Device IP Address: 10.248.203.184
[WiFi] Device Subnet Address: 255.254.0.0
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[Hypnos] Interrupt successfully attached!
[Hypnos] Current Time: 2022.08.25 10:33:05
[Hypnos] Next Interrupt Alarm Set For: 2033.08.28 00:00:15

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Chime",
    "instance": 4
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.263574123
      }
    },
    {
      "module": "SHT31",
      "data": {
        "Temperature": 24.95999908,
        "Humidity": 53.31000137
      }
    },
    {
      "module": "TSL2591",
      "data": {
        "Visible": 62566,
        "Infrared": 873,
        "Full Spectrum": 3159
      }
    },
    {
      "module": "GS3_0",
      "data": {
        "Temperature": 0,
        "Dielectric Permittivity": 0,
        "Conductivity": 0
      }
    },
    {
      "module": "WiFi",
      "data": {
        "SSID": "OSU_Access",
        "RSSI": -57
      }
    }
  ],
  "timestamp": {
    "time_utc": "2106-02-06 6:28:16",
    "time_local": "2106-02-06 6:28:16"
  }
}

[SD Manager] Successfully logged data to Chime5.csv
[MQTT] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[MQTT] Successfully connected to broker!
[MQTT] Attempting to send data...
[MQTT] Data has been successfully sent!
[Hypnos] Interrupt successfully reattached!
[Hypnos] Entering Standby Sleep...

udellc, 2022-08-25:
I tried swapping the hypnos for a different one. It booted up just fine and started taking measurements, but I forgot to put the MQTT creds on the SD, so I powered it off, put the creds on the SD, and now this one hangs too.  I think there is something the M0 is telling the RTC that is causing the erratic behavior in the most recent release, or some releases back since I updated last

WL-Richards, 2022-08-25:
> After reprogramming with the Hypnos_RTC_CustomTime example, removing and reinserting a new battery, I am getting prompted. Unclear if it is because reprogramming with the example and power cycling the RTC helped, or a new battery.

You likely didn't take the battery out for long enough, this should be done when the device is powered off, pull the battery for 10 seconds or so treat it like resetting a cmos

WL-Richards, 2022-08-25:
Ahhh I might know hang tight

WL-Richards, 2022-08-26:
Addressed by removing useRTC flag from Hypnos

closed 2022-08-26 by WL-Richards