OPEnSLab-OSU/Loom-V4 #23: Pin interrupt issue waking M0 from sleep and solution

udellc, 2022-09-01:
The current implementation of the SparkFun LowPower library only allows pin interrupts of type LOW.
Working with this type, attaching and detaching interrupts can be a hassle and confusing to many just getting started.
Working with Falling, rising, and change would be easier and much more flexible in the variety of trigger types to wake the M0 from sleep.

This was not an issue in Loom 3 and below because of all the abstraction of the Digital module and Interrupts module in old Loom. However, we are asking users now to manage their own pin interrupts and async timers for transparency. BUT it is making it harder to figure out how to wake the M0 beyond the Hypnos baked in interrupt functionality for the RTC which is stable.

**Describe the solution you'd like**
LowPower library from sparkfun does not have rising, falling, and change features patched. But 
https://www.arduino.cc/reference/en/libraries/arduino-low-power/
Arduino Low Power library does. This has been tested on the Feather M0 in OPEnS Lab.
I would like to propose changing sleep functions over to Arduino Low Power library.

**Additional context**
Dependency for the Library is RTCZero Library.

WL-Richards, 2022-09-07:
Implemented on separate branch because changes are monumental, needs testing

WL-Richards, 2022-09-09:
ArduinoLowPower with fully customizable interrupts has been implemented to main

closed 2022-09-09 by WL-Richards