OPEnSLab-OSU/Loom-V4 #26: Be able to set time zone from SD JSON parameter

udellc, 2022-09-06:
**Is your feature request related to a problem? Please describe.**
Loom-programmed devices can end up in different time zones after programming, but the only way to set the timezone is during programming. We can currently set sample period via a file on the SD Card. Is it possible to add an option timezone field in the JSON for people to set it? Timezone should still default to the programmed time zone if the SD card info is not found , not defined, or null.

udellc, 2022-09-06:
Also, When the User is prompted to enter Time/Date Information via the Serial Monitor in the current functionality, the user is not prompted for TimeZone. This can confuse the UTC time if the user does not know what timezone the device was originally programmed.

WL-Richards, 2022-09-07:
Added ability to load timezone from SD

closed 2022-09-07 by WL-Richards