OPEnSLab-OSU/Loom-V4 #27: Important: Need 2 MS5803 sensors on different i2c addr to log at same time

udellc, 2022-09-08:
**Describe the bug**
Attempting to use 2 MS5803 sensors. 
Needed to calculate water height, using one pressure sensor above water, the other below. Pressure difference translates to water height.
MS5803 data sheet page 8 allows 2 MS5803 sensors to be used on same system by tying CSB pin to VCC or GND in hardware to make different i2c addresses. VCC=0x76 i2c addr ; GND=0x77 addr

I made a sketch testing using 2, hardware is correct. When Loom Logs, it only logs the last MS5803 read, instead of logging both as unique sensors. 

Both MS5803 sensors seem to be successfully initialized in the printout.

**Hardware in Use**
Hypnos, Feather M0 WiFi, 2 MS5803 sensors with CSB pins configured properly.

**To Reproduce**
Load included code with above hardware.

**Expected behavior**
2 distinct MS5803 sensors need to be logged seperately. 
There also needs to be a way to differentiate clearly WHICH 5803 the data column belongs to; e.g. 
MS5803_77 and MS5803_76

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab test example of using 2 MS5803 sensors each set to different address</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses a hypnos, MS5803 with CBS pin on GND and another MS5803 with CBS pin on VCC</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;WC_MS_test&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_2</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Sensors to use</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms77</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">119</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; MS5803 CSB pin pulled GND i2c addr 0x77</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms66</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">118</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; MS5803 CSB pin pulled VCC i2c addr 0x76</font>

<font color="#434f54">&#47;&#47; Might be desirable to pull sleep interval from SD card, not relevant to sketch, disregard</font>
<font color="#000000">TimeSpan</font> <font color="#000000">sleepInterval</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; Sleep interval pulled from JSON file on SD Card, see setup</font>

<font color="#434f54">&#47;&#47; ISR not used in loop, no sleep in use</font>
<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">wakeup</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Wait 20 seconds for the serial console to open</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Enable the hypnos rails</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">enable</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">sleepInterval</font> <font color="#434f54">=</font> <font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">getSleepIntervalFromSD</font><font color="#000000">(</font><font color="#005c5f">&#34;SD_config.json&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#434f54">&#47;&#47; Register the ISR and attach to the interrupt</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">registerInterrupt</font><font color="#000000">(</font><font color="#000000">isrTrigger</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure and package the data</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
[... 126 lines cut ...]
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 3
      }
    },
    {
      "module": "MS5803",
      "data": {
        "Temperature": 23.72999954,
        "Pressure": 1012.599976
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-08-26 0:0:33",
    "time_local": "2022-08-25 17:0:33"
  }
}

[SD Manager] Successfully logged data to WC_MS_test7.csv

**Additional context**
Add any other context about the problem here.

WL-Richards, 2022-09-08:
This was an oversight, will implement asap

WL-Richards, 2022-09-09:
Should be working and ready for you to test

udellc, 2022-09-09:
Both sensors now show up in the packet and are uniquely identifiable.
However, When one sensor or the other is used by itself, I get accurate data and the generic print of MS5803 is shown on the serial monitor - all good. 
When I use both, The last sensor has accurate data, but the first does not.

Print with one sensor or the other:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to WC_MS_test13.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
Oversampling setting: 512
C0 = 16
C1 = 42299
C2 = 37213
C3 = 26721
C4 = 25242
C5 = 31870
C6 = 27618
C7 = 23242
p_crc: 202
n_crc: 10
[MS5803] Successfully Initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Sleep interval successfully loaded from SD!
[Hypnos] Registering RTC interrupt...

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "WC_MS_test",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "MS5803",
      "data": {
        "Temperature": 25.42000008,
        "Pressure": 995.4199829
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-08-27 3:54:41",
    "time_local": "2022-08-26 20:54:41"
  }
}

[SD Manager] Successfully logged data to WC_MS_test13.csv

Print from using just the other sensor (still not both):
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to WC_MS_test14.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
Oversampling setting: 512
C0 = 36
C1 = 39571
C2 = 37127
C3 = 25726
C4 = 24863
C5 = 32993
C6 = 28679
C7 = 14613
p_crc: 21
n_crc: 5
[MS5803] Successfully Initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Sleep interval successfully loaded from SD!
[Hypnos] Registering RTC interrupt...

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "WC_MS_test",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "MS5803",
      "data": {
        "Temperature": 24.20000076,
        "Pressure": 995.4400024
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-08-27 3:55:48",
    "time_local": "2022-08-26 20:55:48"
  }
}

NOW
Print with both sensors:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to WC_MS_test12.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
Oversampling setting: 512
C0 = 36
C1 = 39571
C2 = 37127
C3 = 25726
C4 = 24863
C5 = 32993
C6 = 28679
C7 = 14613
p_crc: 21
n_crc: 5
[MS5803_119] Successfully Initialized!
Oversampling setting: 512
C0 = 16
C1 = 42299
C2 = 37213
C3 = 26721
C4 = 25242
C5 = 31870
C6 = 27618
C7 = 23242
p_crc: 202
n_crc: 10
[MS5803_118] Successfully Initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Sleep interval successfully loaded from SD!
[Hypnos] Registering RTC interrupt...

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "WC_MS_test",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "MS5803_119",
      "data": {
        "Temperature": 33.72000122,
        "Pressure": 1185.530029
      }
    },
    {
      "module": "MS5803_118",
      "data": {
        "Temperature": 25.18000031,
        "Pressure": 995.5
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-08-27 3:52:2",
    "time_local": "2022-08-26 20:52:2"
  }
}

[SD Manager] Successfully logged data to WC_MS_test12.csv

--
This cannot be true because the temp does not jump to 90ish degrees f in here nor the pressure.
What up?

WL-Richards, 2022-09-09:
That's weird but idk if that's a code thing on our end wouldn't make any sense considering those are two separate modules, I'll look into it though

WL-Richards, 2022-09-09:
Is it possible 119 is bad cause 118 are the values shown both times

udellc, 2022-09-14:
119 works on its own, and so does 118; but both together, the first one logged gets its data thrown off. I am assuming they are logging one at a time, each address being queried separately. 
119 and 118 values are each shown in the example where both of them are used above and pasted again below.

[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to WC_MS_test12.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
Oversampling setting: 512
C0 = 36
C1 = 39571
C2 = 37127
C3 = 25726
C4 = 24863
C5 = 32993
C6 = 28679
C7 = 14613
p_crc: 21
n_crc: 5
[MS5803_119] Successfully Initialized!
Oversampling setting: 512
C0 = 16
C1 = 42299
C2 = 37213
C3 = 26721
C4 = 25242
C5 = 31870
C6 = 27618
C7 = 23242
p_crc: 202
n_crc: 10
[MS5803_118] Successfully Initialized!
[Manager] ** Setup Complete **
[Hypnos] Sleep interval successfully loaded from SD!
[Hypnos] Registering RTC interrupt...

[Manager] Data Json:
{
"type": "data",
"id": {
"name": "WC_MS_test",
"instance": 1
},
"contents": [
{
"module": "Packet",
"data": {
"Number": 1
}
},
{
"module": "MS5803_119",
"data": {
"Temperature": 33.72000122,
"Pressure": 1185.530029
}
},
{
"module": "MS5803_118",
"data": {
"Temperature": 25.18000031,
"Pressure": 995.5
}
}
],
"timestamp": {
"time_utc": "2022-08-27 3:52:2",
"time_local": "2022-08-26 20:52:2"
}
}

[SD Manager] Successfully logged data to WC_MS_test12.csv

WL-Richards, 2022-09-19:
Required to initialize each sensor everytime before measuring

closed 2022-09-19 by WL-Richards