OPEnSLab-OSU/Loom-V4 #32: Max31856 Loom not working, raw example code does

udellc, 2022-09-12:
**Describe the bug**
Max 31856 raw example (oneShot.ino) gives good data on serial monitor.
 Loom version prints errors and bad reading.

This is the example code that works:
<pre>
<font color="#434f54">&#47;&#47; Basic example using one-shot measurement.</font>
<font color="#434f54">&#47;&#47; The call to readThermocoupleTemperature() is blocking for O(100ms)</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Adafruit_MAX31856</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Use software SPI: CS, DI, DO, CLK</font>
<font color="#434f54">&#47;&#47;Adafruit_MAX31856 maxthermo = Adafruit_MAX31856(10, 11, 12, 13);</font>
<font color="#434f54">&#47;&#47; use hardware SPI, just pass in the CS pin</font>
<font color="#000000">Adafruit_MAX31856</font> <font color="#000000">maxthermo</font> <font color="#434f54">=</font> <font color="#000000">Adafruit_MAX31856</font><font color="#000000">(</font><font color="#000000">10</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">115200</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><font color="#434f54">!</font><b><font color="#d35400">Serial</font></b><font color="#000000">)</font> <font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">10</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;MAX31856 thermocouple test&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">maxthermo</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">maxthermo</font><font color="#434f54">.</font><font color="#000000">setThermocoupleType</font><font color="#000000">(</font><font color="#000000">MAX31856_TCTYPE_K</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">&#34;Thermocouple type: &#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#5e6d03">switch</font> <font color="#000000">(</font><font color="#000000">maxthermo</font><font color="#434f54">.</font><font color="#000000">getThermocoupleType</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_B</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;B Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_E</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;E Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_J</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;J Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_K</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;K Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_N</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;N Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_R</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;R Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_S</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;S Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_TCTYPE_T</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;T Type&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_VMODE_G8</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;Voltage x8 Gain mode&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">case</font> <font color="#000000">MAX31856_VMODE_G32</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;Voltage x8 Gain mode&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#5e6d03">default</font><font color="#434f54">:</font> <b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;Unknown&#34;</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#5e6d03">break</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">}</font>

<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">&#34;Cold Junction Temp: &#34;</font><font color="#000000">)</font><font color="#000000">;</font>
[... cut ...]
<font color="#95a5a6"> * Pass in a variable number of arguments to the construct to designate the number of samples and which pin you&#39;re using</font>
[... 48 lines cut ...]
[MAX31856] Over/Under Voltage Fault
[MAX31856] Thermocouple Open Fault

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "MAX31856",
      "data": {
        "Temperature": null
      }
    }
  ]
}

udellc, 2022-09-13:
Document fix please!

WL-Richards, 2022-09-13:
changed name of variable from max to maxthermo?

udellc, 2022-09-14:
That might do it as max looks to be reserved in the IDE

udellc, 2024-12-03:
Elliot found a way to use hardware spi with this chip. This uses less GPIO. can we implement it? But let's first check to ensure there are no strange things with the SD card. SPI functionality. There does seem to be an issue when reading the time period for sampling from the SD card using hardware spy, but it doesn't seem to be an issue. Any use software spy for the max chip.

sarveshta, 2025-01-21:
Elliot will test

WL-Richards, 2025-01-22:
This is on branch max31856-hardware-spi

sarveshta, 2025-01-24:
I'm assigning a review of this fix to you @ZimHockey13. Contact Will/me if you have any questions.

ZimHockey13, 2025-02-05:
Testing the new fix for the max31856, we are still encountering the get config from SD card sleep interval error; details are shown below. The setup uses a Hypnos board, which uses SPI communication for the SD card reader, and a K-Type thermocouple, which uses the MAX31856 board and also relies on SPI communication.

sd card config file:
`
{
    "SleepInterval": {
        "days" : 0,
        "hours" : 0,
        "minutes" : 0,
        "seconds" : 30
    },
    "timezone": "AKST"
}
`

serial output when asked to serialize the config json:
`
{"":{"days":0,"hours":0,"minutes":0,"seconds":30},"timezone":"AKST"}[2025.02.04 16:56:45] [DEBUG] [:getConfigFromSD:582] Config successfully loaded from SD!
[2025.02.04 16:56:45] [DEBUG] [:getConfigFromSD:586] Selected timezone: AKST, UTC offset: -9
[2025.02.04 16:56:45] [ERROR] [:getConfigFromSD:597] There was an error reading the sampling interval from SD, defaulting sampling interval to 20 minutes.
`

We see that the "timezone" header was successfully read and the correct timezone information was implemented. However, when the SPI thermocouple device is connected the header for "SleepInterval" is deleted or corrupted to "" which is why the sampling interval cannot be read from SD. After removing the SPI thermocouple sensor, the SD card is read correctly with no header dropping. This leads me to believe there is some deeper conflict with the SPI devices that we will have to investigate.

sarveshta, 2025-02-11:
I had a similar issue when helping Quinn with his onboarding exam. He was trying to use a sensor (I don't recall which, but it was not SPI) and LoRa. The same issue happened. I will look into this as well with Quinn's setup.

sarveshta, 2025-02-11:
moving ``free(fileRead);`` to the very end of the ``getConfigFromSD`` right before ``FUNCTION_END;`` makes it work!

```
TimeSpan Loom_Hypnos::getConfigFromSD(const char* fileName){
    FUNCTION_START;
    // Doc to store the JSON data from the SD card in
    StaticJsonDocument<OUTPUT_SIZE> doc;
    char output[OUTPUT_SIZE];
    char* fileRead = sdMan->readFile(fileName);
    DeserializationError deserialError = deserializeJson(doc, fileRead);

    // Create json object to easily pull data from
    JsonObject json = doc.as<JsonObject>();

    if(deserialError != DeserializationError::Ok){
        snprintf(output, OUTPUT_SIZE, "There was an error reading the config from SD: %s, defaulting sampling interval to 20 minutes.", deserialError.c_str());
        ERROR(output);
        return TimeSpan(0, 0, 20, 0);
    }
    else{
        LOG(F("Config successfully loaded from SD!"));
        if(!json["timezone"].isNull()){
            const char* timezoneStr = json["timezone"].as<const char*>();
            snprintf(output, OUTPUT_SIZE, "Selected timezone: %s, UTC offset: %i", timezoneStr, (int)timezoneMap[(const char*)timezoneStr]);
            LOG(output);
            timezone = timezoneMap[(const char*)timezoneStr];
        }

        // If the sleep interval key is not supplied we want to set some default
        if(!json["SleepInterval"].isNull()){
            // Return the interval as set in the json
            return TimeSpan(json["SleepInterval"]["days"].as<int>(), json["SleepInterval"]["hours"].as<int>(), json["SleepInterval"]["minutes"].as<int>(), json["SleepInterval"]["seconds"].as<int>());
        }
        else{
            snprintf(output, OUTPUT_SIZE, "There was an error reading the sampling interval from SD, defaulting sampling interval to 20 minutes.");
            ERROR(output);
            return TimeSpan(0, 0, 20, 0);
        }
    }
    free(fileRead);
    FUNCTION_END;
}
```

sarveshta, 2025-02-21:
Assigned to @quinnedgar. I will shadow his testing.

quinnedgar, 2025-02-21:
Tested with Sarvesh, code hanging on [DEBUG] [:initialize:217] ** Initializing Modules ** 

Used Adafruit example code for MAX31856_oneshot.ino to test wiring and boards functionality. MAX31856 correctly reports values with Adafruit code using hardware SPI constructor instead of software SPI constructor. 

---------------------------
Uploaded Adafruit example:

// Basic example using one-shot measurement.
// The call to readThermocoupleTemperature() is blocking for O(100ms)

#include <Adafruit_MAX31856.h>

// Use software SPI: CS, DI, DO, CLK
//Adafruit_MAX31856 maxthermo = Adafruit_MAX31856(10, 11, 12, 13);
// use hardware SPI, just pass in the CS pin
Adafruit_MAX31856 maxthermo = Adafruit_MAX31856(10);

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(10);
  Serial.println("MAX31856 thermocouple test");

  maxthermo.begin();

  maxthermo.setThermocoupleType(MAX31856_TCTYPE_K);

  Serial.print("Thermocouple type: ");
  switch (maxthermo.getThermocoupleType() ) {
    case MAX31856_TCTYPE_B: Serial.println("B Type"); break;
    case MAX31856_TCTYPE_E: Serial.println("E Type"); break;
    case MAX31856_TCTYPE_J: Serial.println("J Type"); break;
    case MAX31856_TCTYPE_K: Serial.println("K Type"); break;
    case MAX31856_TCTYPE_N: Serial.println("N Type"); break;
    case MAX31856_TCTYPE_R: Serial.println("R Type"); break;
    case MAX31856_TCTYPE_S: Serial.println("S Type"); break;
    case MAX31856_TCTYPE_T: Serial.println("T Type"); break;
    case MAX31856_VMODE_G8: Serial.println("Voltage x8 Gain mode"); break;
    case MAX31856_VMODE_G32: Serial.println("Voltage x8 Gain mode"); break;
    default: Serial.println("Unknown"); break;
  }

}

void loop() {
  Serial.print("Cold Junction Temp: ");
  Serial.println(maxthermo.readCJTemperature());

  Serial.print("Thermocouple Temp: ");
  Serial.println(maxthermo.readThermocoupleTemperature());
  // Check and print any faults
  uint8_t fault = maxthermo.readFault();
  if (fault) {
    if (fault & MAX31856_FAULT_CJRANGE) Serial.println("Cold Junction Range Fault");
    if (fault & MAX31856_FAULT_TCRANGE) Serial.println("Thermocouple Range Fault");
    if (fault & MAX31856_FAULT_CJHIGH)  Serial.println("Cold Junction High Fault");
    if (fault & MAX31856_FAULT_CJLOW)   Serial.println("Cold Junction Low Fault");
    if (fault & MAX31856_FAULT_TCHIGH)  Serial.println("Thermocouple High Fault");
    if (fault & MAX31856_FAULT_TCLOW)   Serial.println("Thermocouple Low Fault");
    if (fault & MAX31856_FAULT_OVUV)    Serial.println("Over/Under Voltage Fault");
    if (fault & MAX31856_FAULT_OPEN)    Serial.println("Thermocouple Open Fault");
  }
  delay(1000);
}

With above code, program correctly polls MAX31856 and reads as
15:16:17.488 -> Cold Junction Temp: 24.11
15:16:17.488 -> Thermocouple Temp: 22.58
15:16:18.640 -> Cold Junction Temp: 24.19
15:16:18.640 -> Thermocouple Temp: 22.60

---------------------
Uploaded Loom code:

/**
 * Temperature Readings using teh MAX31856 sensor
 * Pass in a variable number of arguments to the construct to designate the number of samples and which chip pins you're using for SPI
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */

#include <Loom_Manager.h>

#include <Sensors/SPI/Loom_MAX318XX/Loom_MAX31856.h>

Manager manager("Device", 1);

// create sensor
// default parameters:
// Manager& man, int samples = 1, int chip_select = 10, int mosi = -1, int miso = -1, int sclk = -1, TEMP_UNIT unit = CELCIUS
Loom_MAX31856 maxthermo(manager);

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Initialize the manager
  manager.initialize();
}

void loop() {
  // put your main code here, to run repeatedly:

  // Measure the data from the sensors
  manager.measure();

  // Package the data into JSON
  manager.package();

  // Print the JSON document to the Serial monitor
  manager.display_data();

  // Wait for 5 seconds
  manager.pause(5000);
}

This code hangs on DEBUG message: 

 15:12:42.522 -> [DEBUG] [:initialize:217] ** Initializing Modules **

WIRING:

MAX31856       Feather M0
     VIN          <--   3V
    GND          <--   GND
    SCK          <--   SCK
     SDO         <--   MI
    SDI           <---   MO
    CS          <--   10

quinnedgar, 2025-02-28:
Code tested with Sarvesh with following .ino off max31856-hardware-spi, works when tested with Hypnos. 

/**
 * Temperature Readings using teh MAX31856 sensor
 * Pass in a variable number of arguments to the construct to designate the number of samples and which chip pins you're using for SPI
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */

#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/SPI/Loom_MAX318XX/Loom_MAX31856.h>

Manager manager("Device", 1);
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

// create sensor
// default parameters:
// Manager& man, int samples = 1, int chip_select = 10, int mosi = -1, int miso = -1, int sclk = -1, TEMP_UNIT unit = CELCIUS
Loom_MAX31856 maxthermo(manager);

void setup() {

  // Start the serial interface
  manager.beginSerial();
  hypnos.enable();

  // Initialize the manager
  manager.initialize();
}

void loop() {
  // put your main code here, to run repeatedly:

  // Measure the data from the sensors
  manager.measure();

  // Package the data into JSON
  manager.package();

  // Print the JSON document to the Serial monitor
  manager.display_data();

  // Wait for 5 seconds
  manager.pause(5000);
}

rainewheary, 2025-04-18:
Note: revisit around week 6 if no fires, @ZimHockey13 (Elliott) can assist.