OPEnSLab-OSU/Loom-V4 #33: LTE 4G shield cannot connect to network

udellc, 2022-09-13:
**Describe the bug**
Attempting to use the 4G LTE board from Will to connect WeatherChimes sensors to internet for MongoDB uploads.
Shield powers up and communicates with M0, but does not connect to network.

BTW: I had to switch the hardware switches on the 4G board to "Hardware SPI" and power from "Shield." Both switches were switched to the other sides (software spi and power from Arduino)
Board would not work in those original switched configurations, but setting to the new positions got the board power and communication.
What was the reason for that?

I notice the example on Loom also has an arduino_secrets.h tab, but I thought we transitioned to an SD card JSON file for this. Am I running out of date code?

One last issue; the program hangs if LET cannot find the internet signal - It should instead carry on and log minimally to SD, and also allow WiFi to be used if declared as needing to be used, as a backup (e.g. if a wifi M0 is used on the 4G board, we should also be able to enable WiFi logging too as a just-in-case backup if we want).

**Hardware in Use**
Weatherchimes PCB, M0, SaraR4 4G board, TSL2590, MS5803, GS3
Hypnos

**To Reproduce**
Run below code, taken from WeatherChimes 4G example on Loom. 

**Expected behavior**
Should connect to internet and log sensors to MongoDB

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create the TSL2591 and SHT classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_SDI12</font> <font color="#000000">sdi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MS5803</font> <font color="#d35400">ms</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">119</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; 119(0x77) if CSB=LOW default, 118(0x76) if CSB=HIGH</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color [... cut ...]
[MS5803] Successfully Initialized!
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[SHT31] Failed to initialize SHT31! Check connections and try again...
[TSL2591] Successfully initialized TSL2591!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[LTE] Not initialized!

**Additional context**
Add any other context about the problem here.

WL-Richards, 2022-09-13:
The power switch needs to be set to arduino as per discussed in the Meeting not LTE
As for the hanging there isn't much that can be done about that as that is simply how the LTE module works
Change the power switch back to ARDUINO and try again reopen if issue presists

udellc, 2022-09-13:
Switched the power switch to Arduino and still getting the following:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Chime55.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[LTE] Powering up GPRS Modem. This should take about 10 seconds...
[LTE] Powering up complete!
[LTE] Modem Information: ATI  Manufacturer: u-blox Model: SARA-R410M-02B Revision: L0.0.00.00.05.08 [Apr 17 2019 19:34:02] SVN: 03 IMEI: 355523762522961
[LTE] Waiting for network...
[LTE] No Response from network!
[LTE] Module failed to initialize
Oversampling setting: 512
C0 = 36
C1 = 39571
C2 = 37127
C3 = 25726
C4 = 24863
C5 = 32993
C6 = 28679
C7 = 14613
p_crc: 21
n_crc: 5
[MS5803] Successfully Initialized!
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[SHT31] Successfully initialized SHT31!
[TSL2591] Successfully initialized TSL2591!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[LTE] Not initialized!

udellc, 2022-09-13:
Also, Is there an example code to initialize and configure the 4G boards from a feather M0? The example code on how to do this is not clear because they use software serial and don't use the M0 serial calls. We should include this configure 4G board sketch in the Loom examples.

WL-Richards, 2022-09-13:
`/*
  Register your LTE Shield/SIM combo on a mobile network operator
  By: Jim Lindblom
  SparkFun Electronics
  Date: November 19, 2018
  License: This code is public domain but you buy me a beer if you use this 
  and we meet someday (Beerware license).
  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/14997

  This example demonstrates how to initialize your Cat M1/NB-IoT shield, and
  connect it to a mobile network operator (Verizon, AT&T, T-Mobile, etc.).

  Before beginning, you may need to adjust the mobile network operator (MNO)
  setting on line 45. See comments above that line to help select either
  Verizon, T-Mobile, AT&T or others.

  You may also need to set an APN on line 51 -- e.g. "hologram"
  
  Hardware Connections:
  Attach the SparkFun LTE Cat M1/NB-IoT Shield to your Arduino
  Power the shield with your Arduino -- ensure the PWR_SEL switch is in
    the "ARDUINO" position.
*/

//Click here to get the library: http://librarymanager/All#SparkFun_LTE_Shield_Arduino_Library
#include <SparkFun_LTE_Shield_Arduino_Library.h>

// Create a LTE_Shield object to be used throughout the sketch:
LTE_Shield lte;

// To support multiple architectures, serial ports are abstracted here.
// By default, they'll support AVR's like the Arduino Uno and Redboard
// For example, on a SAMD21 board SerialMonitor can be changed to SerialUSB
// and LTEShieldSerial can be set to Serial1 (hardware serial port on 0/1)
#define SerialMonitor Serial
#define LTEShieldSerial Serial1

// Network operator can be set to either:
// MNO_SW_DEFAULT -- DEFAULT
// MNO_ATT -- AT&T 
// MNO_VERIZON -- Verizon
// MNO_TELSTRA -- Telstra
// MNO_TMO -- T-Mobile
const mobile_network_operator_t MOBILE_NETWORK_OPERATOR = MNO_ATT;
const String MOBILE_NETWORK_STRINGS[] = {"Default", "SIM_ICCD", "AT&T", "VERIZON", 
  "TELSTRA", "T-Mobile", "CT"};

// APN -- Access Point Name. Gateway between GPRS MNO
// and another computer network. E.g. "hologram
const String APN = "hologram";

// This defines the size of the ops struct array. Be careful making
// this much bigger than ~5 on an Arduino Uno. To narrow the operator
// list, set MOBILE_NETWORK_OPERATOR to AT&T, Verizeon etc. instead
// of MNO_SW_DEFAULT.
#define MAX_OPERATORS 5

#define DEBUG_PASSTHROUGH_ENABLED

void setup() {
  int opsAvailable;
  struct operator_stats ops[MAX_OPERATORS];
  String currentOperator = "";
  bool newConnection = true;

  // Turn on the LTE Board
  pinMode(5, OUTPUT);                     // 3.3v power rail
  pinMode(6, OUTPUT);                     // 5v power rail
  pinMode(A5, OUTPUT);
[... 138 lines cut ...]
  for (int i = 0; i < operatorsAvailable; i++) {
    SerialMonitor.print(String(i + 1) + ": ");
    SerialMonitor.print(ops[i].longOp + " (" + String(ops[i].numOp) + ") - ");
    switch (ops[i].stat) {
    case 0:
      SerialMonitor.println(F("UNKNOWN"));
      break;
    case 1:
      SerialMonitor.println(F("AVAILABLE"));
      break;
    case 2:
      SerialMonitor.println(F("CURRENT"));
      break;
    case 3:
      SerialMonitor.println(F("FORBIDDEN"));
      break;
    }
  }
  SerialMonitor.println();
}

void serialWait() {
  while (!SerialMonitor.available()) ;
  while (SerialMonitor.available()) SerialMonitor.read();
}`

udellc, 2022-09-14:
Running the above code; the LET shield hangs. See prints below:
Initializing the LTE Shield...
...this may take ~25 seconds if the shield is off.
...it may take ~5 seconds if it just turned on.
LTE Shield connected!

udellc, 2022-09-14:
... Do you think the 1.0.6 update has something potentially to do with the issue? I can try 1.0.5 tomorrow, but maybe worth a look?

WL-Richards, 2022-09-14:
I don't think the update should have broken anything

WL-Richards, 2022-09-15:
PCB was not functioning properly

closed 2022-09-15 by WL-Richards