OPEnSLab-OSU/Loom-V4 #34: Loom Max breaking change on 4.0.6 but works on 4.0.5

udellc, 2022-09-13:
**Describe the bug**
Loom Max does not work in 4.0.6 but does in 4.0.5

**Hardware in Use**
M0 WiFi and Wattson board.

**To Reproduce**
Upload included code in 4.0.5 to see working, then same in 4.0.6 to see bug.

**Expected behavior**
4.0.6 build and code should work for Loom Max

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for using the MPU gyroscope in conjunction with max</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Wifi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Communication</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Max</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MPU6050</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MPU6050</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Digital</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Digital</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Actuators</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Neopixel</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Neopixel</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Wattson&#34;</font><font color="#434f54">,</font> <font color="#000000">11</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47;Loom_WIFI wifi(manager, CommunicationMode::CLIENT, SECRET_SSID, SECRET_PASS); &#47;&#47; Use this line if connection should be in Client mode using router SSID and password from arduino_secrets.h tab</font>
<font color="#000000">Loom_WIFI</font> <font color="#000000">wifi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">CommunicationMode</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">AP</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; Use this line if WiFi connection should be in Access Point mode</font>

<font color="#000000">Loom_Max</font> <font color="#000000">maxMsp</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">,</font> <font color="#00979c">new</font> <font color="#000000">Loom_Neopixel</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; register neopixel and wifi to the manager</font>
<font color="#000000">Loom_MPU6050</font> <font color="#000000">mpu</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Read the battery voltage and A0 and A1</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A0</font><font color="#434f54">,</font> <font color="#000000">A1</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Reads the button on pin 10</font>
<font color="#000000">Loom_Digital</font> <font color="#000000">digital</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#434f54">,</font> <font color="#000000">9</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Send and Recieve data from Max</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#000000">publish</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#d35400">subscribe</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">pause</font><font color="#000000">(</font><font color="#000000">40</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>
**Config**
arduino_secrets.h tab has
#define SECRET_SSID "SSID"
#define SECRET_PASS "Password"

**4.0.5 Output**

[Manager] Initializing Modules...
[WiFi] Initializing WiFi module...
[WiFi] Starting access point on: Wattson11
[... 157 lines cut ...]
        "az": 1.078125,
        "gx": -10.03211975,
        "gy": 54.866642,
        "gz": 7.798936844,
        "roll": -250.9406128,
        "pitch": 1372.463013,
        "yaw": 199.0600586
      }
    },
    {
      "module": "WiFi",
      "data": {
        "SSID": "Wattson2",
        "RSSI": -100
      }
    }
  ]
}

[Max Pub/Sub] Sending packet to 0.0.0.255:8002
[Max Pub/Sub] An error occurred when attempting to close the current packet!
[Max Pub/Sub] No message received!

**Additional context**
Add any other context about the problem here.

WL-Richards, 2022-09-13:
I didn't change anything between the 2 regarding that but ill look

WL-Richards, 2022-09-13:
Possible fix

udellc, 2022-09-13:
It may have something to do with the low power library change?
The printouts are very different, even the order in which things are initialized; e.g. the wifi initializes before the gyro in old version. Gyro before wifi in new

udellc, 2022-09-13:
Is the fix in Main?
for testing?

WL-Richards, 2022-09-13:
> 

Doubt its the library
And yes the fix is in main

WL-Richards, 2022-09-13:
Wait I know why

WL-Richards, 2022-09-13:
It has to do with it sorting the keys in the map

WL-Richards, 2022-09-13:
Should be fixed, this may have been causing additional weird behavior as well

closed 2022-09-13 by WL-Richards

udellc, 2022-09-13:
Great! I'll test

udellc, 2022-09-15:
Fix Confirmed on AP and Client!