OPEnSLab-OSU/Loom-V4 #35: MUX TCA9548A does not initalize

udellc, 2022-09-13:
**Describe the bug**
MUX example for Loom Sensors does not function. TCA9548A does not initalize.

**Hardware in Use**
Feather M0 and Multiplexer breakout board for TCA9548A

**To Reproduce**
Connect hardware, Attach an i2c sensor to one of the 8 ports, or don't - the device should report which sensors, if any, are on which MUX channel.

**Expected behavior**
MUX should initialize in the serial printouts. Report sensors plugged into the MUX board and on which channel.

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * Uses the I2C Multiplexer to dynamically allow hot swapping of I2C sensors</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Multiplexer</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Multiplexer</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Reads the battery voltage</font>
<font color="#000000">Loom_Multiplexer</font> <font color="#000000">mux</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Start the serial interface</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Initialize the manager</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure the data from the sensors</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Package the data into JSON</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Print the JSON document to the Serial monitor</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#434f54">&#47;&#47; put your main code here, to run repeatedly:</font>

<font color="#000000">}</font>

</pre>
**Config**
In the Arduino IDE while editing the config.json, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

**Output**
[Manager] Initializing Modules...
[Manager] ** Setup Complete ** 

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    }
  ]
}

**Additional context**
Add any other context about the problem here.

udellc, 2022-09-13:
Attempted same code with Hypnos added to manager and powered on. This helps for some i2c sensors that have not yet been fixed. Hypnos initializes, but still no MUX, so issue is probably not related to the issue with some of the other i2c sensors:

Code:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * Uses the I2C Multiplexer to dynamically allow hot swapping of I2C sensors</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Multiplexer</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Multiplexer</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_2</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Reads the battery voltage</font>
<font color="#000000">Loom_Multiplexer</font> <font color="#000000">mux</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Start the serial interface</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">enable</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Initialize the manager</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure the data from the sensors</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Package the data into JSON</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Print the JSON document to the Serial monitor</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#434f54">&#47;&#47; put your main code here, to run repeatedly:</font>

<font color="#000000">}</font>

</pre>

Printout:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Device0.csv
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[Manager] ** Setup Complete ** 

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-09-13 21:15:56",
    "time_local": "2022-09-13 14:15:56"
  }
}

WL-Richards, 2022-09-13:
I fixed the mux issue it is in the integration-hell branch
It just hadn't been pushed to main

closed 2022-09-13 by WL-Richards