OPEnSLab-OSU/Loom-V4 #37: RTC won't register interrupt when using GSM shield WeatherChimes Integration Hell

udellc, 2022-09-16:
**Describe the bug**
RTC won't register the interrupt when using the GSM shield on the WeatherChimes 4G code on Integration Hell branch
Order of initialization of components looks suspicious on printouts.

**Hardware in Use**
Hypnos V3_2, WiFi M0, GSM board, Weatherchimes PCB, I left other sensor unplugged.
Note I get a different printout and order when I use the Hypnos V3_3. See comment below

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code**
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_2</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create the TSL2591 and SHT classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_SDI12</font> <font color="#000000">sdi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A0</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms_water</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">119</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; 119(0x77) if CSB=LOW external, 118(0x76) if CSB=HIGH on WC PCB</font>
<font color="#434f54">&#47;&#47;Loom_MS5803 ms_air(manager, 118); &#47;&#47; 118(0x76) if CSB=HIGH on WC PCB</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color="#434f54">,</font> <font color="#000000">BROKER_USER</font><font color="#434f54">,</font> <font color="#000000">BROKER_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
[... cut ...]
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

[LTE] Module successfully initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...

**Additional context**
Add any other context about the problem here.

udellc, 2022-09-16:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create the TSL2591 and SHT classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
[... cut ...] p;&nbsp;&nbsp;&nbsp;

[... 50 lines cut ...]
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.361865044,
        "Vbat_MV": 4324.804688
      }
    },
    {
      "module": "TSL2591",
      "data": {
        "Visible": 0,
        "Infrared": 0,
        "Full_Spectrum": 0
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-09-16 19:20:4",
    "time_local": "2022-09-16 12:20:4"
  }
}

[SD Manager] Successfully logged data to Chime0.csv
[MQTT] Creating new MQTT client!

udellc, 2022-09-16:
Note, the one above, it looks like it tries to create an MQTT client after the first sample is already taken...

udellc, 2022-09-16:
I am using a V2.3 Hypnos board, but with traces cut to reroute connections to the latest board.

udellc, 2022-09-16:
This one is a legit bug I think. I can run the striped down WeatherChimes logging only analog on 4G,
But, it gets hanged at registering the RTC interrupt if I enable SDI12 on A0:
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Chime7.csv
[Hypnos] RTC lost power, lets set the time!
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[SDI12] Initializing SDI-12 Sensors...
[SDI12] Scanning SDI-12 Address Space this make take a little while...
[SDI12] == We found the following active Addresses ==
[SDI12]     Address: 0
[LTE] Powering up GPRS Modem. This should take about 10 seconds...
[LTE] Powering up complete!
[LTE] Modem Information: Manufacturer: u-blox Model: SARA-R410M-02B Revision: L0.0.00.00.05.08 [Apr 17 2019 19:34:02] SVN: 03 IMEI: 355523762522953
[LTE] Waiting for network...
[LTE] Connected to network!
[LTE] Attempting to connect to LTE Network: hologram
[LTE] Successfully Connected!
[LTE] Connected!
[LTE] APN: hologram
[LTE] Signal State: 99
[LTE] IP Address: 100.71.133.128
[LTE] Attempting to verify internet connection...
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Fri, 16 Sep 2022 21:47:22 GMT
Content-Type: text/plain; charset=UTF-8
Content-Length: 121
Connection: close
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

[LTE] Module successfully initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...

udellc, 2022-09-16:
Issue lies with SDI12 module. 
Stripped version of WeatherChimes 4G with only analog and SHT31 works. 
Stripped version of WeatherChimes 4G with only analog, TSL2591, and SHT31 works.

WL-Richards, 2022-09-16:
> Note, the one above, it looks like it tries to create an MQTT client after the first sample is already taken...

This is a non-issue

udellc, 2022-09-16:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47;#include &lt;Sensors&#47;SDI12&#47;Loom_SDI12&#47;Loom_SDI12.h&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47;#include &lt;Sensors&#47;I2C&#47;Loom_MS5803&#47;Loom_MS5803.h&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create sensor classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47;Loom_SDI12 sdi(manager, A0);</font>
<font color="#434f54">&#47;&#47;Loom_MS5803 ms(manager, 119); &#47;&#47; 119(0x77) if CSB=LOW external, 118(0x76) if CSB=HIGH on WC PCB</font>
<font color="#434f54">&#47;&#47;Loom_MS5803 ms_air(manager, 118); &#47;&#47; 118(0x76) if CSB=HIGH on WC PCB</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font> [... cut ...]
[MQTT] Creating new MQTT client!
[MQTT] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[MQTT] Successfully connected to broker!
[MQTT] Attempting to send data...
[MQTT] Data has been successfully sent!
[Hypnos] Current Time: 2022.09.16 11:43:54
[Hypnos] Next Interrupt Alarm Set For: 2022.09.16 11:58:54
[Hypnos] Interrupt successfully reattached!
[LTE] Powering down GPRS Modem. This should take about 5 seconds...
[LTE] Powering down complete!
[Hypnos] Entering Standby Sleep...

udellc, 2022-09-16:
Now with SDI12 stuff uncommented:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47;#include &lt;Sensors&#47;I2C&#47;Loom_MS5803&#47;Loom_MS5803.h&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create sensor classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
[... cut ...]
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

[LTE] Module successfully initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...

udellc, 2022-09-16:
SDI12 example  works with only M0 and GS3 hooked up.
Will add Hypnos 3_3 up and see

udellc, 2022-09-16:
Adding Hypnos 3_3 into hardware and code works:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for the SDI-12 implementation WITH the use of a manger to allow for maximum usability</font>
<font color="#95a5a6"> * This example scans the address space for devices and the requests data from the first device it finds</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Manager handles all loom simplicity </font>
<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Device&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_SDI12</font> <font color="#000000">sdi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A0</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">wakeup</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Start the serial interface</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Enable the hypnos rails</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">enable</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Initialize the manager</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Register the ISR and attach to the interrupt</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">registerInterrupt</font><font color="#000000">(</font><font color="#000000">isrTrigger</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#434f54">&#47;&#47; put your main code here, to run repeatedly:</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure the data from the sensors</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Package the data into JSON</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Print the JSON document to the Serial monitor</font>
[... cut ...]
    },
    {
      "module": "GS3_0",
      "data": {
        "Temperature": 0,
        "Dielectric_Permittivity": 0,
        "Conductivity": 0
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-09-16 18:43:43",
    "time_local": "2022-09-16 11:43:43"
  }
}

[SD Manager] Successfully logged data to Device0.csv
[Hypnos] Current Time: 2022.09.16 11:43:43
[Hypnos] Next Interrupt Alarm Set For: 2022.09.16 11:43:53
[Hypnos] Interrupt successfully reattached!

[Hypnos] Entering Standby Sleep...

udellc, 2022-09-16:
May have something to do with the AREF pin on the 4G shield... but I don't know how.

udellc, 2022-09-16:
Okay, So the code hangs here when I have the M0 and Hypnos attached to the 4G board. See code and prints:

<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for the SDI-12 implementation WITH the use of a manger to allow for maximum usability</font>
<font color="#95a5a6"> * This example scans the address space for devices and the requests data from the first device it finds</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Manager handles all loom simplicity </font>
<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_SDI12</font> <font color="#000000">sdi</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A0</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color="#434f54">,</font> <font color="#000000">BROKER_USER</font><font color="#434f54">,</font> <font color="#000000">BROKER_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">wakeup</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

[... cut ...]
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

[LTE] Module successfully initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...

udellc, 2022-09-16:
Strange thing is, if it were an AREF tied to ground issue on the 4G board preventing the A0 connection; then why can I still use the analog module to read and log battery voltage? This is successful!?!?

<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for the SDI-12 implementation WITH the use of a manger to allow for maximum usability</font>
<font color="#95a5a6"> * This example scans the address space for devices and the requests data from the first device it finds</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47;#include &lt;Sensors&#47;SDI12&#47;Loom_SDI12&#47;Loom_SDI12.h&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Manager handles all loom simplicity </font>
<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A2</font><font color="#434f54">,</font> <font color="#000000">A4</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47;Loom_SDI12 sdi(manager, A0);</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color="#434f54">,</font> <font color="#000000">BROKER_USER</font><font color="#434f54">,</font> <font color="#000000">BROKER_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
[... cut ...]
[MQTT] Creating new MQTT client!
[MQTT] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[MQTT] Successfully connected to broker!
[MQTT] Attempting to send data...
[MQTT] Data has been successfully sent!
[Hypnos] Current Time: 2022.09.16 15:52:21
[Hypnos] Next Interrupt Alarm Set For: 2022.09.16 15:57:21
[Hypnos] Interrupt successfully reattached!
[LTE] Powering down GPRS Modem. This should take about 5 seconds...
[LTE] Powering down complete!
[Hypnos] Entering Standby Sleep...

udellc, 2022-09-16:
Tried setting analog pin to output will break the hang:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This is an example use case for the SDI-12 implementation WITH the use of a manger to allow for maximum usability</font>
<font color="#95a5a6"> * This example scans the address space for devices and the requests data from the first device it finds</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><b><font color="#d35400">SDI12</font></b><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SDI12</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Manager handles all loom simplicity </font>
<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;Chime&#34;</font><font color="#434f54">,</font> <font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#434f54">&#47;&#47;Loom_Analog analog(manager, A2, A4);</font>
<font color="#434f54">&#47;&#47;Loom_SDI12 sdi(manager, A2);</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color="#434f54">,</font> <font color="#000000">BROKER_USER</font><font color="#434f54">,</font> <font color="#000000">BROKER_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
[... cut ...]
[MQTT] Creating new MQTT client!
[MQTT] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[MQTT] Successfully connected to broker!
[MQTT] Attempting to send data...
[MQTT] Data has been successfully sent!
[Hypnos] Current Time: 2022.09.16 16:04:44
[Hypnos] Next Interrupt Alarm Set For: 2022.09.16 16:09:44
[Hypnos] Interrupt successfully reattached!
[LTE] Powering down GPRS Modem. This should take about 5 seconds...
[LTE] Powering down complete!
[Hypnos] Entering Standby Sleep...

WL-Richards, 2022-09-17:
Had to do with the LTE module trying to use the Analog pin and declaring the input mode after the SDI-12 device had already tried to do this, the pinMode should also be set before measure to ensure this doesn't happen

closed 2022-09-17 by WL-Richards