OPEnSLab-OSU/Loom-V4 #44: Possible breaking change: 2 MS5803 sensors i2c

udellc, 2022-09-30:
Pulled the main 9/30/22 at 4pm.
using two MS5803 sensors worked in version before in integration hell branch. 
Possible changes to i2c NAK failing gracefully may have caused break in the two MS5803 0x118 snd 0x119 functionality. 
Shows both MS5803s initializing, but only one reports, does not say which one.

Tested in Weatherchimes code below:
<pre>
<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab use case example for the WeatherChimes project</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses SDI12, TSL2591 and an SHT31 sensor to log environment data and logs it to both the SD card and also MQTT&#47;MongoDB</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;arduino_secrets.h&#34;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Analog</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">&#47;</font><font color="#000000">Loom_SHT31</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">&#47;</font><font color="#000000">Loom_TSL2591</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47; If using Teros 10, Uncoment this line</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">Analog</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Teros10</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Teros10</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; If using SDI12, GS3 or Teros 11 or 12 uncoment this line</font>
<font color="#434f54">&#47;&#47;#include &lt;Sensors&#47;SDI12&#47;Loom_SDI12&#47;Loom_SDI12.h&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Logging</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MQTT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Internet</font><font color="#434f54">&#47;</font><font color="#000000">Connectivity</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">&#47;</font><font color="#000000">Loom_LTE</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;4Gchime&#34;</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#434f54">,</font> <font color="#00979c">true</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Analog for reading battery voltage</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create sensor classes</font>
<font color="#000000">Loom_SHT31</font> <font color="#000000">sht</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_TSL2591</font> <font color="#000000">tsl</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms_water</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">119</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; 119(0x77) if CSB=LOW external, 118(0x76) if CSB=HIGH on WC PCB</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms_air</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">118</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; 118(0x76) if CSB=HIGH on WC PCB</font>

<font color="#000000">Loom_LTE</font> <font color="#000000">lte</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">NETWORK_APN</font><font color="#434f54">,</font> <font color="#000000">NETWORK_USER</font><font color="#434f54">,</font> <font color="#000000">NETWORK_PASS</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">Loom_MQTT</font> <font color="#000000">mqtt</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">lte</font><font color="#434f54">.</font><font color="#000000">getClient</font><font color="#000000">(</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">SECRET_BROKER</font><font color="#434f54">,</font> <font color="#000000">SECRET_PORT</font><font color="#434f54">,</font> <font color="#000000">DATABASE</font><font color="#434f54">,</font> <font color="#000000">BROKER_USER</font><font color="#434f54">,</font> <font color="#000000">BROKER_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; If using Teros 10, Uncoment this line</font>
<font color="#000000">Loom_Teros10</font> <font color="#000000">t10</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A2</font><font color="#000000">)</font><font color="#000000">;</font>

[... cut ...]
[MQTT] Creating new MQTT client!
[MQTT] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[MQTT] Successfully connected to broker!
[MQTT] Attempting to send data...
[MQTT] Data has been successfully sent!
[Hypnos] Current Time: 2022.09.30 16:39:39
[Hypnos] Next Interrupt Alarm Set For: 2022.09.30 16:54:39
[Hypnos] Interrupt successfully reattached!
[LTE] Powering down GPRS Modem. This should take about 5 seconds...
[LTE] Powering down complete!
[Hypnos] Entering Standby Sleep...

jamopopper, 2022-10-03:
Do you know what version and branch you were using when it did work? The commit ID would be great, but branch and/or time you pulled previously works too @udellc

udellc, 2022-10-03:
I used Will's push to Main he did on Friday during our Loom meeting.

udellc, 2022-10-07:
Not sure if any progress has been made. Just pulled from Main again just now and still not getting both sensors. Maybe code to fix was reverted? Because it is behaving like last time.

[SD Manager] Initializing SD Card...
[SD Manager] Failed to Initialize SD Card! SD Card functionality will be disabled, is there an SD card inserted into the device?
[Hypnos] DS3231 Real-Time Clock Initialized Successfully!
[Manager] Initializing Modules...
[MS5803] Successfully Initialized!
[MS5803] Successfully Initialized!
[Manager] ** Setup Complete ** 
[Hypnos] Registering RTC interrupt...
[Hypnos] Interrupt successfully attached!

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "WC_MS_test",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "MS5803",
      "data": {
        "Temperature": 24.77000046,
        "Pressure": 1007.190002
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-10-07 0:7:44",
    "time_local": "2022-10-06 17:7:44"
  }
}

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "WC_MS_test",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    },
    {
      "module": "MS5803",
      "data": {
        "Temperature": 24.76000023,
        "Pressure": 1007.169983
      }
    }
  ],
  "timestamp": {
    "time_utc": "2022-10-07 0:7:51",
    "time_local": "2022-10-06 17:7:51"
  }
}

<pre>

[... 66 lines cut ...]
<font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&#47;&#47; Add height to the package</font>
<font color="#95a5a6"> &nbsp;manager.addData(&#34;MS5803both&#34;, &#34;WaterHeight_mm&#34;, height);</font>
<font color="#95a5a6">*&#47;</font>
 &nbsp;<font color="#434f54">&#47;&#47; Print the current JSON packet</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;

 &nbsp;<font color="#434f54">&#47;&#47; Log the data to the SD card &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.logToSD();</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">pause</font><font color="#000000">(</font><font color="#000000">5000</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; If Using Sleep, uncomment:</font>

 &nbsp;<font color="#434f54">&#47;&#47; Set the RTC interrupt alarm to wake the device in 10 seconds</font>
 &nbsp;<font color="#434f54">&#47;&#47;hypnos.setInterruptDuration(sleepInterval); &#47;&#47; if using sd card sleep interval</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 10)); &#47;&#47; if not using sd card sleep interval</font>

 &nbsp;<font color="#434f54">&#47;&#47; Reattach to the interrupt after we have set the alarm so we can have repeat triggers</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.reattachRTCInterrupt();</font>
 &nbsp;
 &nbsp;<font color="#434f54">&#47;&#47; Put the device into a deep sleep, operation HALTS here until the interrupt is triggered</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.sleep(false);</font>

<font color="#000000">}</font>

</pre>

WL-Richards, 2022-10-07:
Fixed

closed 2022-10-07 by WL-Richards