OPEnSLab-OSU/Loom-V4 #46: Loom Max Connectivity, transition from Client to AP mode

udellc, 2022-10-05:
For Max Wifi connectivity - When devices are programmed in Client Mode to join a Wifi router - After 10 tries to connect to router have failed, the device should revert to AP Mode and broadcast an SSID based on their name and instance number and connect as if in AP mode. The current code does not make this transition. See serial prints and code below:
[Manager] Initializing Modules...
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: LoomHub4
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Failed to connect to the access point after 10 tries! Is the network in range and are your credentials correct?
[WiFi] Verifying Connection to the Internet...
[WiFi] Ping Failed! Error Code: Unknown_Host
[WiFi] Successfully Initalized Wifi!
[WiFi] Device IP Address: 0.0.0.0
[WiFi] Device Subnet Address: 0.0.0.0
[Max Pub/Sub] Initializing Max Communication....
[Max Pub/Sub] Listening for UDP Packets on 9000
[Max Pub/Sub] Connections Opened!
[Max Pub/Sub] Initializing desired actuators...
[Neopixel0] Succsessfully initialized Neopixel
[Max Pub/Sub] Successfully initialized actuators!
[MPU6050] Calibrating Gyroscope...
[MPU6050] 
========================================
Calculating gyro offsets
DO NOT MOVE MPU6050...
Done!
X : -10.57
Y : -0.87
Z : -2.49
Program will start after 3 seconds
========================================
[MPU6050] Successfully initialized sensor!
[Manager] ** Setup Complete ** 

[Manager] Data Json: 
{
  "type": "data",
  "id": {
    "name": "Wattson",
    "instance": 0,
    "ip": [
      0,
      0,
      0,
      0
    ]
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "WiFi",
      "data": {
        "SSID": null,
        "RSSI": -100
      }
    },
    {
      "module": "MPU6050",
      "data": {
[... 58 lines cut ...]
<font color="#000000">Loom_Max</font> <font color="#000000">maxMsp</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">wifi</font><font color="#434f54">,</font> <font color="#00979c">new</font> <font color="#000000">Loom_Neopixel</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; register neopixel and wifi to the manager</font>
<font color="#000000">Loom_MPU6050</font> <font color="#000000">mpu</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Read the battery voltage and A0 and A1</font>
<font color="#000000">Loom_Analog</font> <font color="#000000">analog</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">A0</font><font color="#434f54">,</font> <font color="#000000">A1</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#434f54">&#47;&#47; Reads the button on pin 10</font>
<font color="#000000">Loom_Digital</font> <font color="#000000">digital</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#434f54">,</font> <font color="#000000">9</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">display_data</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Send and Recieve data from Max</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#000000">publish</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">maxMsp</font><font color="#434f54">.</font><font color="#d35400">subscribe</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">pause</font><font color="#000000">(</font><font color="#000000">40</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

WL-Richards, 2022-10-07:
Wasn't switching to AP mode when the access point had a password

udellc, 2022-10-08:
Tried the latest pull and all I get is this on the serial monitor:
[Manager] Initializing Modules...
[WiFi] Initializing WiFi module...
[WiFi] Attempting to connect to SSID: LoomHub4
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Attempting to connect to AP...
[WiFi] Failed to connect to the access point after 10 tries! Is the network in range and are your credentials correct?
[WiFi] Attempting to connect to AP...
[WiFi] Failed to connect to the access point after 10 tries! Is the network in range and are your credentials correct?
[WiFi] Attempting to connect to AP...
[WiFi] Failed to connect to the access point after 10 tries! Is the network in range and are your credentials correct?
[WiFi] Attempting to connect to AP...

WL-Richards, 2022-10-08:
Was missing a return should work now

closed 2022-10-08 by WL-Richards