OPEnSLab-OSU/Loom-V4 #47: Set target router SSID, Password from Max feature needs fixed

udellc, 2022-10-05:
From Max, it is useful to be able to connect to a Loom device via Access Point Mode, and send it information to save new SSID and Password info and switch to client mode. Loom device should from then on try to connect to a router with the SSID and password info until it receives a new message to update the info. The Loom device should always go back to AP mode after 10 failed attempts to connect to a router in a row.

Command sent from Max looks like:
print: MaxSub c Kalani UdellFamilyWireless
print: MaxSub c Kalani UdellFamilyWireless
js: [0]  Cmd :   MaxSub c Kalani UdellFamilyWireless   
js: {"type":"command","dst":{"name":"test","instance":0},"commands":[{"module":"MaxSub","func":99,"params":["Kalani","UdellFamilyWireless"]}]}  

But there is no acknowledgement of receipt of this command on the Arduino Serial monitor.

Can this feature be added back?

udellc, 2022-10-07:
This capability was added in Loom3 to the AP Wifi Flash Config branch:
https://github.com/OPEnSLab-OSU/Loom/tree/ap_wifi_flash_config
The reason this was branched I think is because a new flash memory feature was added that was experimental, which saved changes to the "config" to flash memory space; so settings could be changed, saved, and retrieved on next power cycle.

udellc, 2022-10-07:
[SubscribePlats/Max_Sub.cpp](https://github.com/OPEnSLab-OSU/Loom/commit/463f144743401a93ebd2b982b62e61572a3ae956#diff-93de4c8228a5fcc72d4dacda0fa23eee76f27df14dcb8298fc88922c88af7646)

WL-Richards, 2022-10-12:
Implemented needs testing, reopen for future integration issues

udellc, 2022-10-12:
Tried testing after pulling from Main; No change in behavior. 
No reply from M0 serial monitor any message was ever received. 
Max reports sending this command:
print: MaxSub c Kalani UdellFamilyWireless
js: [0]  Cmd :   MaxSub c Kalani UdellFamilyWireless   
js: {"type":"command","dst":{"name":"test","instance":0},"commands":[{"module":"MaxSub","func":99,"params":["myRouter","myPassword"]}]}

WL-Richards, 2022-10-12:
It would appear I committed but forgot to push

WL-Richards, 2022-10-13:
pushed

closed 2022-10-13 by WL-Richards