OPEnSLab-OSU/Loom-V4 #49: Need MS5803 sensor examples

udellc, 2022-10-12:
We need an example of using 1 ms5803, and another example of using 2 of these with the different addresses (118, 119)
Here's some very messy code I was working on using both, please clean up to only the necessary things.

<pre>

<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * In lab test example of using 2 MS5803 sensors each set to different address</font>
<font color="#95a5a6"> * </font>
<font color="#95a5a6"> * This project uses a hypnos, MS5803 with CBS pin on GND and another MS5803 with CBS pin on VCC</font>
<font color="#95a5a6"> * Calculate Water Height using difference of 2 pressure sensors and formulas from:</font>
<font color="#95a5a6"> * </font><u><font color="#95a5a6">https:&#47;&#47;byjus.com&#47;water-pressure-formula&#47;</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"> * (Pwater-Pair)&#47;0.0981 = water height in millimeters</font>
<font color="#95a5a6"> * Convert millibar into Pascals: 1Pa = 0.01 millibar</font>
<font color="#95a5a6"> * </font><u><font color="#95a5a6">https:&#47;&#47;www.unitconverters.net&#47;pressure&#47;pascal-to-millibar.htm</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"> * MANAGER MUST BE INCLUDED FIRST IN ALL CODE</font>
<font color="#95a5a6"> *&#47;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom_Manager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#434f54">&#47;&#47;#include &lt;Wire.h&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Hardware</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">&#47;</font><font color="#000000">Loom_Hypnos</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Sensors</font><font color="#434f54">&#47;</font><font color="#000000">I2C</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">&#47;</font><font color="#000000">Loom_MS5803</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#00979c">float</font> <font color="#000000">height</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; in millimeters</font>
<font color="#00979c">float</font> <font color="#000000">coeficient</font> <font color="#434f54">=</font> <font color="#000000">10.1936</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; take reciprocal of 0.0981 so we can multiply instead of divide &nbsp;(100mBar * 1000mm) &#47; (1000kg.over.metercube * 9.81meterspersecondsquare)</font>
<font color="#00979c">float</font> <font color="#000000">Pw</font> <font color="#434f54">=</font> <font color="#000000">1</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; Pressure water in millibar</font>
<font color="#00979c">float</font> <font color="#000000">Pa</font> <font color="#434f54">=</font> <font color="#000000">1</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; Pressure air in millibar</font>

<font color="#000000">Manager</font> <font color="#000000">manager</font><font color="#000000">(</font><font color="#005c5f">&#34;WC_MS_test&#34;</font><font color="#434f54">,</font> <font color="#000000">1</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Create a new Hypnos object</font>
<font color="#000000">Loom_Hypnos</font> <font color="#000000">hypnos</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">HYPNOS_VERSION</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">V3_3</font><font color="#434f54">,</font> <font color="#000000">TIME_ZONE</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">PST</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Sensors to use</font>
<font color="#434f54">&#47;&#47; Pressure of Water</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms77</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">119</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; MS5803 CSB pin pulled GND i2c addr 0x77</font>

<font color="#434f54">&#47;&#47; Pressure of Air</font>
<font color="#000000">Loom_MS5803</font> <font color="#000000">ms76</font><font color="#000000">(</font><font color="#000000">manager</font><font color="#434f54">,</font> <font color="#000000">118</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47; MS5803 CSB pin pulled VCC i2c addr 0x76</font>

<font color="#434f54">&#47;&#47; Might be desirable to pull sleep interval from SD card, not relevant to sketch, disregard</font>
<font color="#434f54">&#47;&#47;TimeSpan sleepInterval; &#47;&#47; Sleep interval pulled from JSON file on SD Card, see setup</font>

<font color="#434f54">&#47;&#47; ISR not used in loop, no sleep in use</font>
<font color="#434f54">&#47;&#47; Called when the interrupt is triggered </font>
<font color="#00979c">void</font> <font color="#000000">isrTrigger</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#000000">wakeup</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Wait 20 seconds for the serial console to open</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">beginSerial</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">&#47;&#47; Enable the hypnos rails</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">enable</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">initialize</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">&#47;&#47; &nbsp;sleepInterval = hypnos.getSleepIntervalFromSD(&#34;SD_config.json&#34;);</font>
 &nbsp;<font color="#434f54">&#47;&#47; Register the ISR and attach to the interrupt</font>
 &nbsp;<font color="#000000">hypnos</font><font color="#434f54">.</font><font color="#d35400">registerInterrupt</font><font color="#000000">(</font><font color="#000000">isrTrigger</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>

 &nbsp;<font color="#434f54">&#47;&#47; Measure and package the data</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">manager</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#95a5a6">&#47;*</font>
<font color="#95a5a6"> &nbsp;&#47;&#47; Calculate water height</font>
<font color="#95a5a6"> &nbsp;Pw = ms77.getPressure();</font>
<font color="#95a5a6"> &nbsp;Pa = ms76.getPressure();</font>
[... 1 lines cut ...]
<font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&#47;&#47; Add height to the package</font>
<font color="#95a5a6"> &nbsp;manager.addData(&#34;MS5803both&#34;, &#34;WaterHeight_mm&#34;, height);</font>
<font color="#95a5a6">*&#47;</font>
[... cut ...]
<font color="#434f54">&#47;&#47; &nbsp;hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 10)); &#47;&#47; if not using sd card sleep interval</font>

 &nbsp;<font color="#434f54">&#47;&#47; Reattach to the interrupt after we have set the alarm so we can have repeat triggers</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.reattachRTCInterrupt();</font>
 &nbsp;
 &nbsp;<font color="#434f54">&#47;&#47; Put the device into a deep sleep, operation HALTS here until the interrupt is triggered</font>
<font color="#434f54">&#47;&#47; &nbsp;hypnos.sleep(false);</font>

<font color="#000000">}</font>

</pre>

WL-Richards, 2022-10-12:
Do you want the water height calculations?

WL-Richards, 2022-10-12:
Added a two MS5803 example

udellc, 2022-10-12:
I'd like the water height calculation worked out with Cara and tested on the 2 x sensor version of the example code please.

WL-Richards, 2022-10-23:
Added in recent commit

closed 2022-10-23 by WL-Richards