OPEnSLab-OSU/Loom-V4 #61: Important integrations to Loom to expand into Hydrological community

udellc, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
We want to increase the offerings of this system to a broader range of communities and scientists. Specifically those studying water. There are currently-existing sensor products and libraries that are widely used in the hydrological community. We'd like to integrate support for these in the Loom system.

**Describe the solution you'd like**
We should identify key sensors that have already been integrated in open-source systems like this one, and integrate and test:
https://github.com/EnviroDIY/ModularSensors#supported-sensors-

We should also identify key sensors and capabilities that current open systems do not offer to distinguish our system from others.

**Describe alternatives you've considered**
SmartRock was OPEnS hydro-oriented solution, and may continue to be to some extent. However, WeatherChimes is much more capable and workshop and STEAM education friendly, with more formal curriculum and documentation developed. Integrating a broader range of products into the Loom system would increase the audiences we can apply our solutions towards.

**Additional context**
Add any other context or screenshots about the feature request here.

udellc, 2022-12-02:
The Yosemitech and Atlas sensor libraries are in particular the first ones we should try to integrate

udellc, 2023-01-03:
Viable sensors that do not require significant or any voltage level shifting, e.g. 3V compatible are:
https://atlas-scientific.com/probes/co2-sensor/
https://atlas-scientific.com/embedded-solutions/ezo-dissolved-oxygen-circuit/
https://atlas-scientific.com/embedded-solutions/ezo-orp-circuit/
- skip the EC product, we have a better 4-pin one we will offer
https://atlas-scientific.com/embedded-solutions/ezo-ph-circuit/

udellc, 2023-01-03:
Looks like Yosemitech: water quality sensors will require at least a 5V power supply, 5V Logic Level Converter, AND RS485 port in order to become compatible.
e.g.
http://en.yosemitech.com/upLoad/down/month_2005/202005111004391887.pdf

udellc, 2023-01-05:
For the Atlas Scientific sensors, we could include this color sensor (good for algae blooms)
https://files.atlas-scientific.com/EZO_RGB_Datasheet.pdf

WL-Richards, 2023-01-07:
Completed Sensors
- [x] EZO Dissolved Oxygen Sensor
- [ ]  EZO CO2 Sensor
- [ ]  EZO ORP Sensor
- [ ] EZO RGB Color Sensor

WL-Richards, 2023-02-18:
All EZO sensors have been integrated, awaiting testing

closed 2023-02-18 by WL-Richards