OPEnSLab-OSU/Loom-V4 #63: Clock time spontaneously set ahead with UTC and local mismatch

carawalter, 2023-01-04:
Rain savor unit 2 time changed from 12/22/2022 1:58:41 to  11/3/2045 1:58:46 for UTC, and from 12/21/2022 17:58:41 to 2022-102-15 17:58:46 for local from packet 2909 to 2910. It did not take any data after that.

Code: https://github.com/OPEnSLab-OSU/RainSavor/blob/main/Code/RainSavor_LoomV4_1_standby/RainSavor_LoomV4_1_standby.ino
[RainSavor2_27.csv](https://github.com/OPEnSLab-OSU/Loom-V4/files/10347245/RainSavor2_27.csv)
[RS2_RTC_error_2023-01-04.txt](https://github.com/OPEnSLab-OSU/Loom-V4/files/10347247/RS2_RTC_error_2023-01-04.txt)

udellc, 2023-01-20:
Hi Cara, is this still happening? Do you think it is hardware related? e.g. the exact code is fine with some hardware, but not with other hardware?

WL-Richards, 2023-02-18:
Closing unless otherwise specified, as this seems stale

closed 2023-02-18 by WL-Richards