OPEnSLab-OSU/Loom-V4 #80: Analog module not reporting correctly

udellc, 2023-07-10:
**Describe the bug**
When using these analog channels (but I assume this is similar with any combination) 
Loom_Analog analog(manager, A0, A2);
I am getting the following data reported:
{
09:09:36.586 ->       "module": "Analog",
09:09:36.586 ->       "data": {
09:09:36.586 ->         "Vbat": 4.307080269,
09:09:36.586 ->         "Vbat_MV": 4311.914063,
09:09:36.586 ->         " ": 1567,
09:09:36.586 ->         "Vbat_MVuM_MV": 1262.783936
09:09:36.586 ->       }
What does uM_MV mean? 
I should be seeing Vbat, A0 and A2 reported, 
Also, for sake of minimizing data, do we need to have both V and MV reported? This is an easy conversion. Let's just pick one, V is fine.

**Hardware in Use**
Trying to get the SmartRock code updated for a CUAHSI visiting researcher this week.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code**

```cpp
/**
 * In lab use case example for the SmartRock project
 * 
 * This project uses a hypnos, an ADS1115 and a MS5803
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */
#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Sensors/I2C/Loom_MS5803/Loom_MS5803.h>
#include <Sensors/Loom_Analog/Loom_Analog.h> // A0 and A2 used in 2020 Version only
#include <Sensors/I2C/Loom_ADS1115/Loom_ADS1115.h> // used in 2021 version only

Manager manager("SmartRock", 1);

// Create a new Hypnos object
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_2, TIME_ZONE::PST);

// Sensors to use
Loom_MS5803 ms(manager, 119);
// If 2020 version, Read the battery voltage, A0 is TDS, A2 is turbidity, 
Loom_Analog analog(manager, A0, A2);

// if 2021 version:
Loom_ADS1115 ads(manager);

TimeSpan sleepInterval;

// Called when the interrupt is triggered 
void isrTrigger(){
  hypnos.wakeup();
  hypnos.enablePowerUp();
}

void setup() {

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable();
  manager.initialize();

  sleepInterval = hypnos.getSleepIntervalFromSD("SD_config.json");
  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);
}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // If 2021 version, Add labeled columns for turbidity and conductivity
  // This can change between SmartRock revisions, ensure your pins are correct before field usage
  manager.addData("Analog Values", "Conductivity", ads.getAnalog(1));
  manager.addData("Analog Values", "Turbidity", ads.getAnalog(2));

  // Print the current JSON packet
  manager.display_data();            

  // Log the data to the SD card              
  hypnos.logToSD();

  // Set the RTC interrupt alarm to wake the device in 10 seconds
  hypnos.setInterruptDuration(sleepInterval);

  // Reattach to the interrupt after we have set the alarm so we can have repeat triggers
  hypnos.reattachRTCInterrupt();
  
  // Put the device into a deep sleep, operation HALTS here until the interrupt is triggered
  hypnos.sleep();
}
```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`

**SD Config .json is here**

```cpp
{
    "days" : 0,
    "hours" : 0,
    "minutes" : 0,
    "seconds" : 10
}
```

WL-Richards, 2023-07-12:
Needs testing but I have refactored the Analog module to use a struct to contain the data within a vector, in addition, the names are now created once in the constructor which should fix the problem that arises when you change a char* resulting in garbage names

WL-Richards, 2023-07-31:
This has been fixed in f66c101

closed 2023-07-31 by WL-Richards