OPEnSLab-OSU/Loom-V4 #83: Loom example MMA8451 accelerometer bug

CaseyJ37, 2023-08-14:
**Describe the bug**
A clear and concise description of what the bug is.
The MMA8451 (accelerometer) example only reports one axis of data instead of all 3, and the parameters being passed into the manager don't line up with the comments.

**Hardware in Use**
A list of all hardware currently in use
Feather M0 board, and an MMA8451 sensor.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'examples'
2. Click on 'MMA8451'
3. Upload
4. Scroll down to 'serial monitor'
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Data for all three sensors should be dispayed.

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

```cpp
/**
 * MMA8451 Example code
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */

#include <Loom_Manager.h>

#include <Sensors/I2C/Loom_MMA8451/Loom_MMA8451.h>

// CHANGE THIS TO WHATEVER PIN YOU HAVE THE MMA INTERRUPT CONNECTED TO
#define INTERRUPT_PIN 10

Manager manager("Device", 1);

// Reads the battery voltage
// Manger Instance, Address, Range, Interrupt Pin, Interrupt Trigger Sensitivity 
Loom_MMA8451 mma(manager, 0x1D, false, MMA8451_RANGE_2_G, INTERRUPT_PIN, 0x10);

void ISR(){
  Serial.println("Movement Detected!!");
}

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Initialize the manager
  manager.initialize();

  // Set the movement ISR
  mma.setISR(ISR);
}

void loop() {
  // put your main code here, to run repeatedly:

  // Measure the data from the sensors
  manager.measure();

  // Package the data into JSON
  manager.package();

  // Print the JSON document to the Serial monitor
  manager.display_data();

  // Wait for 5 seconds
  manager.pause(5000);
}

```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`
\\\\
[DEBUG] [Loom_Manager.cpp:display_data:173] Data Json: 

{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 122
      }
    },
    {
      "module": "MMA8451",
      "data": {
        "X_Acc": 9.691728592,
        "Orien": "Portrait_Down_Front"
      }
    }
  ]
}
\\\\\\\

**Additional context**
Add any other context about the problem here.

WL-Richards, 2023-08-17:
The only one output from the accelerometer was fixed a while ago it's just that a new board profile hasn't been published yet, as for the comments the false is just to say that we are not using a multiplexer to control the device

HarveyBel, 2023-08-21:
Fixed example comment to match.

closed 2023-08-21 by WL-Richards