OPEnSLab-OSU/Loom-V4 #84: Hazelnut LoRa Hub is down

dbzheleva, 2023-08-16:
Hi all,

I just checked on the hazelnut dendrometers and none have reported since 8-11.

The last package in the hub folder is also from the same date.

WL-Richards, 2023-08-17:
Is there more information to go with this cause this doesn't give me very much to go off of?

nicholasqle, 2023-08-22:
I went out to the field to investigate this hub last Friday. I logged the messages the hub gave when it attempted to send a packet to the broker. 
```
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receive:132] Packet Received!
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 1 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 1 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 2 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 2 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 3 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 3 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 4 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 4 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 5 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 5 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 6 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 6 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 7 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 7 / 8
[DEBUG] [Loom_LoRa.cpp:receivePartial:174] Waiting for packet 8 / 8
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
[DEBUG] [Loom_LoRa.cpp:receivePartial:179] Fragment received 8 / 8
[DEBUG] [Loom_Manager.cpp:display_data:173] Data Json: 
{
  "type": "data",
  "id": {
    "name": "HazelnutDendrometer_",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.149169922,
        "Vbat_MV": 4126.611328
      }
    },
    {
      "module": "Teros10",
      "data": {
        "Millivolt_Reading": 1747.106201,
        "Dielectric_Permittivity": 14.71807861,
        "Volumetric_Water_Content": 0.275456399
      }
    },
    {
"module": "SHT31",
      "data": {
        "Temperature": 20.20000076,
        "Humidity": 68.69999695,
        "VPD": 0.740932643
      }
    },
    {
      "module": "LoRa",
      "data": {
        "RSSI": 0
      }
    },
    {
      "module": "AS5311",
      "data": {
        "Alignment": "Green",
        "mag": 1011,
        "pos_raw": 5,
        "pos_avg": 5
      }
    },
    {
      "module": "displacement",
      "data": {
        "um": 0.48828125
      }
    },
    {
      "module": "Button",
      "data": {
        "Pressed?": true
}
    }
  ],
  "timestamp": {
    "time_utc": "2023-08-18T16:0:41Z",
    "time_local": "2023-08-18T9:0:41Z"
  }
}
[SD Manager] Successfully logged data to Hub8.csv
[DEBUG] [Loom_MQTT.cpp:publish:65] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[ERROR] [Loom_MQTT.cpp:publish:70] Failed to connect to broker: CONNECTION_REFUSED
[DEBUG] [Loom_MQTT.cpp:publish:65] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[ERROR] [Loom_MQTT.cpp:publish:70] Failed to connect to broker: CONNECTION_REFUSED
[DEBUG] [Loom_MQTT.cpp:publish:65] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[ERROR] [Loom_MQTT.cpp:publish:70] Failed to connect to broker: CONNECTION_REFUSED
[DEBUG] [Loom_MQTT.cpp:publish:65] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[DEBUG] [Loom_MQTT.cpp:publish:77] Successfully connected to broker!
[DEBUG] [Loom_MQTT.cpp:publish:78] Attempting to send data...
[ERROR] [Loom_MQTT.cpp:publish:92] Failed to close message!
[DEBUG] [Loom_LoRa.cpp:recv:341] Waiting for packet...
```
The hub will usually seem to work for days or a few weeks and it will suddenly stop working. Restarting the hub by disconnecting the power for a few minutes seems to help somewhat at the success rate of sending data. I remember when Connor and I were originally testing the LTE shield back in June, it would only work with one or a few transmissions before crashing requiring a restart. That may be relevant to this issue as well.

WL-Richards, 2023-08-22:
Okay I've seen this issue with the old MQTT broker before its very sporadic and unsure why this is the case, we will be adding support for the OPEnS lab MongoDB database on the new MQTT broker that should hopefully solve this problem.

WL-Richards, 2023-09-02:
We believe this may be due to a poor LTE connection as other dendrometers with the same setup are operating optimally, closing for now...

closed 2023-09-02 by WL-Richards