OPEnSLab-OSU/Loom-V4 #85: The chime on my desk has gotten stuck (didn't freeze)

WL-Richards, 2023-08-22:
The device seems to have gotten stuck in a loop of infinitely powering up all the modules including the LTE board over and over

A normal power up looks like this
```C
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
[DEBUG] [Loom_LTE.cpp:power_up:106] Powering up complete!
[DEBUG] [src/Logger.h:startFunction:225] connect
[DEBUG] [Loom_LTE.cpp:connect:155] Waiting for network...
[DEBUG] [Loom_LTE.cpp:connect:168] Connected to network!
[DEBUG] [Loom_LTE.cpp:connect:172] Attempting to connect to LTE Network: hologram
[DEBUG] [Loom_LTE.cpp:connect:174] Successfully Connected!
[DEBUG] [Loom_Hypnos.cpp:post_sleep:423] Device has awoken from sleep!
```

Where as towards the end of the log file it just does this 
```C
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [src/Logger.h:startFunction:225] initialize
[DEBUG] [Loom_MS5803.cpp:initialize:22] Successfully Initialized!
[DEBUG] [src/Logger.h:startFunction:225] power_up
[DEBUG] [Loom_LTE.cpp:power_up:99] Powering up GPRS Modem. This should take about 10 seconds...
```

I have never seen anything like this before needs to be looked into

WL-Richards, 2023-08-22:
After resetting the board operation resumes normally

udellc, 2023-08-22:
Looks like a brown out issue, low battery. @WL-Richards please put in more details and close

WL-Richards, 2023-08-23:
Attaching full log output 
[DeskChimeBrownOut.log](https://github.com/OPEnSLab-OSU/Loom-V4/files/12420169/DeskChimeBrownOut.log)

While plugged into USB to see if we can replicate it, we are predicting it was a brownout as the voltage was sitting at 3.6v when the device froze when attempting to power on the LTE board. The more considerable power spike likely caused the voltage to dip below the stable threshold and cause weird glitchy behavior. Closing unless the issue arises again.

closed 2023-08-23 by WL-Richards