OPEnSLab-OSU/Loom-V4 #112: SEN55 Packets not publishing

drcrockerjr, 2024-01-08:
**Describe the bug**
When using SEN55 sensor on new Weather Chimes PCB the serial monitor specifies that data is successfully logged to MongoDB, but data doesn't show up in MongoDB.

**Hardware in Use**
Most recent Weather Chimes unit 
SEN55 Air Quality Sensor
SHT31

**To Reproduce**
Steps to reproduce the behavior:
1. Upload SEN55 code while on "SEN55" branch
2. Look at serial monitor output and wait for output from mgtt.publish
4. See MongoDB to look for packet

**Expected behavior**
It is expected that when the serial monitor says that data was successfully sent to MongoDB that you will see the packet in MongoDB.

**Code**
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Logging/Loom_MQTT/Loom_MQTT.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>

Manager manager("Whisp_PROTO", 1);

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Current Sensing
//Loom_ACS712 acs712(manager, A0);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MQTT mqtt(manager, lte.getClient());

void isrTrigger() 
{
  hypnos.wakeup();
}

void setup() {  
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable();

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  //Sets Fan Cleaning Assuming input is in mS (need to check)
  //SEN55->set_cleaning_interval(1000);

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  hypnos.logToSD();

  mqtt.publish();

  hypnos.setInterruptDuration(TimeSpan(0,0,5,0));

  hypnos.reattachRTCInterrupt();

  hypnos.sleep(false, true, false);

  //manager.pause(2000);

}

**Output**
{
  "type": "data",
  "id": {
    "name": "Whisp_PROTO",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 7
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.299023628,
        "Vbat_MV": 4297.412109
      }
    },
    {
      "module": "SEN55",
      "data": {
        "PM1.0": 2.799999952,
        "PM2.5": 2.099999905,
        "PM4.0": 2.599999905,
        "PM10.0": 2.799999952,
        "AmbientHumidity": 35.09000015,
        "AmbientTemperature": 27.09499931,
        "VocIndex": 0,
        "NoxIndex": 0
      }
    },
    {
      "module": "SHT31",
      "data": {
        "Temperature": 22.95000076,
        "Humidity": 49.40000153
      }
    },
    {
      "module": "LTE",
      "data": {
        "RSSI": 99
      }
    }
  ],
  "timestamp": {
    "time_utc": "2024-01-05T5:41:53Z",
    "time_local": "2024-01-04T21:41:53Z"
  }
}

[2024.01.04 21:41:53] [DEBUG] [:log:159] Successfully logged data to Whisp_PROTO0.csv
[2024.01.04 21:41:53] [DEBUG] [:publish:71] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.01.04 21:41:55] [DEBUG] [:publish:83] Successfully connected to broker!
[2024.01.04 21:41:55] [DEBUG] [:publish:84] Attempting to send data...
[2024.01.04 21:41:58] [DEBUG] [:publish:101] Data has been successfully sent!
[2024.01.04 21:41:58] [DEBUG] [:setInterruptDuration:358] Current Time: 2024.01.04 21:41:58
[2024.01.04 21:41:58] [DEBUG] [:setInterruptDuration:361] Next Interrupt Alarm Set For: 2024.01.04 21:46:58
[2024.01.04 21:41:58] [DEBUG] [:reattachRTCInterrupt:162] Interrupt successfully reattached!
[2024.01.04 21:41:58] [DEBUG] [:power_down:126] Powering down GPRS Modem. This should take about 5 seconds...
[2024.01.04 21:42:03] [DEBUG] [:power_down:130] Powering down complete!
[2024.01.04 21:42:03] [DEBUG] [:sleep:374] Entering Standby Sleep...

**Additional context**
 This is happening on the SEN55 branch

WL-Richards, 2024-01-09:
Needs further testing but is likely caused by the '.' in between the PM1.0 etc. measurements.

Found by these logs on the broker server:
0|node-red | 4 Jan 15:10:24 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'
0|node-red | 4 Jan 15:15:52 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'
0|node-red | 4 Jan 15:21:20 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'
0|node-red | 4 Jan 15:26:47 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'
0|node-red | 4 Jan 15:32:15 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'
0|node-red | 4 Jan 15:37:43 - [error] [mongodb out:25de379ab8b6d0e9] Error: key PM1.0 must not contain '.'

ZakaryW, 2024-01-09:
Replaced the dot with an underscore when packaging data on the SEN55 branch. It'll need to be tested to see if it fixed the issue.

ZakaryW, 2024-01-10:
Douglas tested the changes and the issue has been resolved.

closed 2024-01-10 by ZakaryW