OPEnSLab-OSU/Loom-V4 #122: Port manager JSON to proper ArduinoJson usage. Using strings to store arduino (not urgent)

WL-Richards, 2024-01-19:


ZakaryW, 2024-02-11:
So I updated how JSON data is being handled. Now whenever the JSON document is need, deserializeJson is called with the doc and jsonStr variable. Likewise, whenever changes have been made to the doc obj, serializeJson is called with the doc and jsonStr variable. So jsonStr is the json data stored as a cstring and is used for mediating changes to the document. Changes are on this branch --> https://github.com/OPEnSLab-OSU/Loom-V4/tree/json-overhaul

udellc, 2024-07-12:
@WL-Richards , can we merge this with the Wisp, Chimes branch we're currently using - and plan to have @ZakaryW , @SorenEmmons and @Sarvesh-Thiruppathi test this the DAY it MERGES, to confirm if something is broken or if it works?
Let's see some coordination

WL-Richards, 2024-10-25:
Closing this as its done, and has been sidelined

closed 2024-10-25 by WL-Richards