OPEnSLab-OSU/Loom-V4 #126: LoRa Batch first packet sent/received is null packet

nicholasqle, 2024-01-28:
**Describe the bug**
Every time the LoRa Batch transmission is done, the first packet is always null. The rest of the packets are normal.

**Hardware in Use**
Transmitting Side: Feather M0 LoRa, Dendrometer PCB, AS5311
Receiving Side: Feather M0 LoRa, Weatherchimes PCB, SparkFun LTE CAT M1/NB-IoT Shield

**To Reproduce**
Steps to reproduce the behavior:
1. The batch size for LoRa transmission is reached on the transmitting device
2. Observe the serial monitor on the receiving device to see the displayed packets.

**Expected behavior**
The first packet should contain all of the measurements

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here
Transmitting side (Dendrometer):

```cpp
#include <Loom_Manager.h> //4.6
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Hardware/Actuators/Loom_Neopixel/Loom_Neopixel.h>
#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/Analog/Loom_Teros10/Loom_Teros10.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>
#include <Radio/Loom_LoRa/Loom_LoRa.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

#include "AS5311.h"

//////////////////////////
/* DEVICE CONFIGURATION */
//////////////////////////
static const uint8_t NODE_NUMBER = 1;
static const char * DEVICE_NAME = "BlueberryDendrometer_";
////Select one wireless communication option
#define DENDROMETER_LORA
// #define DENDROMETER_WIFI
////These two time values are added together to determine the measurement interval
static const int8_t MEASUREMENT_INTERVAL_MINUTES = 1;
static const int8_t MEASUREMENT_INTERVAL_SECONDS = 0;
static const uint8_t TRANSMIT_INTERVAL = 16; // to save power, only transmit every X measurements
////Use teros 10?
#define DENDROMETER_TEROS10
//////////////////////////
//////////////////////////

// Pins
#define AS5311_CS A3 // 9 for LB version, A3 otherwise
#define AS5311_CLK A5
#define AS5311_DO A4
#define BUTTON_PIN A1 // 11 for LB, A1 otherwise

// Loom
Manager manager(DEVICE_NAME, NODE_NUMBER);
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST); // 3_2 for LB, 3_3 otherwise
// Loom Sensors
Loom_Analog analog(manager);
#ifdef DENDROMETER_TEROS10
Loom_Teros10 teros(manager, A0);
#endif
Loom_SHT31 sht(manager);
Loom_Neopixel statusLight(manager, false, false, true, NEO_GRB); // using channel 2 (physical pin A2). use RGB for through-hole LED devices. GRB otherwise.

// magnet sensor
AS5311 magnetSensor(AS5311_CS, AS5311_CLK, AS5311_DO);

// wireless
[... 486 lines cut ...]
     }
    },
    {
      "module": "displacement",
      "data": {
        "um": 0
      }
    },
    {
      "module": "Button",
      "data": {
        "Pressed?": false
      }
    }
  ],
  "timestamp": {
    "time_utc": "2024-01-28T18:27:50Z",
    "time_local": "2024-01-28T10:27:50Z"
  }
}
[2024.01.28 10:42:59] [DEBUG] [SDManager.cpp:log:159] Successfully logged data to Hub0.csv
``````
**Additional context**
Add any other context about the problem here.
I am not sure if this is expected behavior but Chet wanted me to post this here for future people

WL-Richards, 2024-01-30:
Testing fix on batch-null-packet

WL-Richards, 2024-02-27:
This should be fixed on batch-null-packet but has not been merged into main yet

nicholasqle, 2024-03-01:
The required change of 
```else if (!recvDoc["id"].isNull() && recvDoc["id"]["name"].isNull()){
                return false;
            }```
has not been committed yet to batch-null-packet

WL-Richards, 2024-03-08:
Now it has been pushed

closed 2024-03-08 by WL-Richards