OPEnSLab-OSU/Loom-V4 #131: Auto Gain and Integration Time for TSL2591

WL-Richards, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
We want to automatically adjust the gain and integration values so we always get good values

Here is an example of the VEML utilizing a similar solution
https://github.com/OPEnSLab-OSU/Lilypad/tree/main/Code/Adafruit_VEML7700_Library_Altered
Refer to the autoLux method for the implementation

ZakaryW, 2024-03-06:
I created some good foundational code for this, hopefully it helps. On [tsl-auto-gain](https://github.com/OPEnSLab-OSU/Loom-V4/tree/tsl-auto-gain) branch.

ZakaryW, 2024-03-06:
^ This should ideally be a solid solution, but it should be tested and compared against our expectations.

ZakaryW, 2024-04-05:
We have decided to abandon the TSL implementation in favor of just using veml instead, as it makes more sense.

closed 2024-04-05 by ZakaryW

WL-Richards, 2024-04-15:
The TSL2591 reports 0 when the sensor is likely supposed to be reading max value (this is assumed because the battery voltage jumps up at the same time, likely showing that the device is switching over to complete power off the solar charger)