OPEnSLab-OSU/Loom-V4 #132: Update Doxygen For Hypnos

ZakaryW, 2024-02-20:
Currently the doxygen documentation does not cover the enums that are outside the Hypnos class, but are used within the Hypnos functionality. So to fix this, we need to regenerate the doxygen and ensure it includes these enums.

ZakaryW, 2024-02-27:
Some extra context,

The config file is located in the root directory of the repo, [Doxyfile](https://github.com/OPEnSLab-OSU/Loom-V4/blob/main/Doxyfile). This is the current [Hypnos Documentation](https://openslab-osu.github.io/Loom-V4/class_loom___hypnos.html), the enums which are in the Hypnos header file aren't documented here so we want to change that. It might be worth looking through Doxygen's documentation for [documenting code](https://www.doxygen.nl/manual/docblocks.html) as well.

WL-Richards, 2024-03-08:
You will likely have to change some of the directory and file paths in the Doxyfile as they are absolute paths on my machine

udellc, 2024-07-12:
@ZakaryW , where are we at with this? Can we finish up and close this out?

udellc, 2024-07-12:
Not a high priority though

WL-Richards, 2024-10-25:
closing for now, can re-address later

closed 2024-10-25 by WL-Richards