OPEnSLab-OSU/Loom-V4 #134: Addalogger Sleep Functionality

SorenEmmons, 2024-02-27:
When using the addalogger board specifically the ENGR 102 board setup and attempting to use Hypnos Sleep the program will enter deep sleep and will not wake up when the interrupt is called resulting in it hanging.

WL-Richards, 2024-02-27:
The datalogger with the DS3231 breakout board is not directly supported currently, this was a backlogged issue that hasn't been integrated yet.

udellc, 2024-07-12:
We ought to spend a week this summer investigating this @WL-Richards and @SorenEmmons

SorenEmmons, 2024-07-19:
Nothing needs to be changed with Loom backend, when registering interrupt the sleep interrupt do hypnos.registerInterrupt(isrTrigger, 10, SLEEP); and hypnos.reattachRTCInterrupt(10); as the SQW interrupt pin is connected to pin 10 not pin 12 when made according to the instructions from Chets ENGR 102 you will also need to add a delay at the end of your code that may vary depending on if the sensor needs to power up or calibrate but for something like the STEMMA sensor a 5-second delay works.

closed 2024-07-19 by SorenEmmons