OPEnSLab-OSU/Loom-V4 #135: SEN55 PM1,2.5,4,10 values reporting out of range

drcrockerjr, 2024-02-27:
**Describe the bug**
When running the SEN55 sensor the values reported for the PM 1,2.5,4, and 10 are occuring outside of the defined 0-1000 range defined in the datasheet.

**Hardware in Use**
WheatherChimes/ Whisp

**To Reproduce**
Steps to reproduce the behavior:
1. Upload code provided below for sunning the SEN55 sensor
2. Place flame or smoke source near SEN55 sensor air inlet 
3. Observer PM values outside of 0-1000 range

**Expected behavior**
The PM values should be between 0-1000

**Code**

```cpp
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Whisp_Proto", 1);

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MongoDB mqtt(manager, lte.getClient());
//A batch is logged every 5 minutes, so 12 per hour (12 * 6 = 72) so mqtt will publish at batch size of 72/ every 6 hours
Loom_BatchSD batchSD(hypnos, 72);

void isrTrigger()
{
  hypnos.wakeup();
}

void setup() {
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Set the LTE board to only powerup when a batch is ready to be sent
  lte.setBatchSD(batchSD);

  // Enable the hypnos rails
  hypnos.enable();

  //Time Sync Using LTE 
  hypnos.setNetworkInterface(&lte);

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  hypnos.networkTimeUpdate();

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  // Log the data to the SD
  hypnos.logToSD();

  // Pass in the batchSD to the mqtt obj to check/ publish a batch of data if ready
  mqtt.publish(batchSD);
  //mqtt.publish();

  // Set the interrupt duration for 5 minutes
  hypnos.setInterruptDuration(TimeSpan(0,0,5,0));

  // Reattach the interrupt
  hypnos.reattachRTCInterrupt();

  // Set the hypnos to sleep, but with power still being supplied to the 5v rail (wait for serial when testing from a computer)
  hypnos.sleep(false, true, false);

  hypnos.networkTimeUpdate();
  //manager.pause(2000);
}

```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`

**Additional context**
Add any other context about the problem here.

WL-Richards, 2024-02-27:
Can you additionally post the serial output?

drcrockerjr, 2024-02-27:
I lost the output in my serial monitor. It is only on the SD card of the unit at this point. I can add it after I am able to get a new Whisp up and running as im currently testing the Whisp i used when i encountered that error.

WL-Richards, 2024-05-02:
Closing as this appeared to be an improper use case

closed 2024-05-02 by WL-Richards