OPEnSLab-OSU/Loom-V4 #136: Loom_ADS12303 Hangs at measure

hockerte, 2024-02-27:
While using the Loom_ADS12303 object the program hangs after manager.measure();

**Hardware in Use**
Feather m0
Feather m4 Express
ADS12303 and Breakout Board (I have extra for testing)

**To Reproduce**
Steps to reproduce the behavior:
Run this code

**Expected behavior**
I expect the code to complete measurments.

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

**Output**
Copy and paste the serial output here if possible wrapped in 
```cpp
#include <Loom_Manager.h>
#include <Sensors/SPI/Loom_ADS1232\Loom_ADS1232.h>

Manager manager("Device", 1);

Loom_ADS1232 ads(manager, 10, 0, 1);

void setup() {
  // put your setup code here, to run once:

  manager.beginSerial();
  manager.initialize();

}

void loop() {
  // put your main code here, to run repeatedly:

  ads.measure();
  delay(1000);

}

```

**Additional context**
It appears from my own oscilloscope probing that the clock signal, which is supposed to be oscillated by the feather is held at a constant level.
When I implement the code using the underlying library (ADS1232_Lib) it works fine.

hockerte, 2024-02-27:
Here's the example that works. There's some extra code I wrote to log things to the SD card.

```cpp
//Loom manage must be included first
#include <Loom_Manager.h>

//Including Hypnos (Loom), SD Manager (Loom), and ADS1232 (not Loom) libraries
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Hardware/Loom_Hypnos/SDManager.h>
#include <ADS1232_Lib.h>

//Define ADS12303 PINS
#define pdwn A2
#define dout A1
#define sclk A0

Manager manager("Device", 1);

ADS1232_Lib ads(pdwn, dout, sclk);

/* Note the hypnos and sd manager are different
 * objects because using the hypnos logging functionaliy
 * requires the use of manager.measure which appears broken
 * for the ADS12303. Thus this script uses the ADS12303 library
 * to take measurements directly.
 */
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true, false);
SDManager sd(&manager, 11);

void isrTrigger(){
  hypnos.wakeup();
}

void setup() {
  // put your setup code here, to run once:

  manager.beginSerial();
  hypnos.enable();
  hypnos.registerInterrupt(isrTrigger);
  manager.initialize();

  //set the SD card output pins (we are NOT using the hypnos log functionality)
  pinMode(23, OUTPUT);
  pinMode(24, OUTPUT);
  pinMode(11, OUTPUT);
  sd.begin();

  //These are the calibration parameters for the load cell
  ads.set_offset(0);
  ads.set_scale(1);
  //not sure if this line is neccesary
  ads.power_up();

}

void loop() {
  //set sleep duration
  ads.power_up();
  hypnos.setInterruptDuration(TimeSpan(0, 0, 1, 0));

  //measure ads counts and convert it to a c string
  long raw = ads.units_read(10);
  char raw_str[16];
  ltoa(raw, raw_str,10);

  //Get the current time from hypnos and save it as a c str
  char buf1[20];
  DateTime now = hypnos.getCurrentTime();
  sprintf(buf1, "%02d:%02d:%02d %02d/%02d/%02d",  now.hour(), now.minute(), now.second(), now.day(), now.month(), now.year());

  //Combine the time and measurments strings. Include "," to separate columns
  char date_and_data[64] = {};
  strcat(date_and_data, buf1);
  strcat(date_and_data, ",");
  strcat(date_and_data, raw_str);

  sd.writeLineToFile("Device6.csv", date_and_data);

  ads.power_down();
  hypnos.reattachRTCInterrupt();
  hypnos.sleep();

}

```

WL-Richards, 2024-03-05:
It would appear that the SCLK and the DOUT got flipped

And SCLK = A1 in Loom and DOUT = A0

Try changing that in the loom code and retest to see if it works and we will make the change

hockerte, 2024-03-05:
I changed the pins and got the same issue.
I'm going to try and implement on different hardware to see if that's the issue.

I changed  the Loom_ads1232 initializer as follows:

```cpp
//////////////////////////////////////////////////////////////////////////////////////////////////////
Loom_ADS1232::Loom_ADS1232(Manager& man, int num_samples, long offset, float scale) : Module("ADS1232"), manInst(&man), inst(ADS1232_Lib(A2, A1, A0)) {
    // Set offset, scale, and number of samples
    this->offset = offset;
    this->scale = scale;
    this->num_samples = num_samples;
    // Set pins
    inst.PDWN = A2;
    inst.SCLK = A0;
    inst.DOUT = A1;
    manInst->registerModule(this);
}
//////////////////////////////////////////////////////////////////////////////////////////////////////
```

WL-Richards, 2024-03-06:
You can actually just use addData to add arbitrary data into the complete JSON packet so it does get logged https://openslab-osu.github.io/Loom-V4/class_manager.html#a944ae6ba25646ec601d2682048801734

WL-Richards, 2024-03-08:
Try out this branch https://github.com/OPEnSLab-OSU/Loom-V4/tree/ads1232-measure-hang
I couldn't find much difference from your example and the loom integration but I moved somethings around so give it a shot

hockerte, 2024-04-25:
Appears working. Most likely soemthing was wrong on my end.

closed 2024-04-25 by hockerte