OPEnSLab-OSU/Loom-V4 #138: MongoDB charts from LoRa Batch Uploads does not seem to always match data from SD card and contains unexpected values

nicholasqle, 2024-02-28:
**Describe the bug**
When graphing the data from LoRa batch on MongoDB, we sometimes get weird noise or spikes in the data that do not appear on the SD card. 
For example here I graphed the same exact time period using the data on MongoDB vs data from SD card over the same time period.
[image]
[image]
When looking at individual packets sometimes there it seems that there is an outlier packet with values that are off from the rest of the data. These weird packets also have an extra field under displacement of undefined: null that the rest of the packets do not have.
[image]

**Hardware in Use**
Dendrometer sending LoRa batch, Feather M0 LoRa, Hypnos v3.3.
Hub receiving LoRa batch, Feather M0 LoRa, Hypnos v3.3, 4G LTE Board.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup LoRa batch on Dendrometer with hub uploading to MongoDB.
2. Collect data over a period of time

**Expected behavior**
Data should not contain unexpected values.

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here
Receiving hub:

```cpp
#include "arduino_secrets.h"

#include <Loom_Manager.h> //4.6

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Radio/Loom_LoRa/Loom_LoRa.h>
#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

const unsigned long REPORT_INTERVAL = 1 * 60 * 60 * 1000;

Manager manager("Hub", 0);
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);
Loom_Analog batteryVoltage(manager);
Loom_LoRa lora(manager);
Loom_LTE lte(manager, "hologram", "", "", A5);
Loom_MongoDB mqtt(manager, lte.getClient(), SECRET_BROKER, SECRET_PORT, DATABASE, BROKER_USER, BROKER_PASS);
Loom_BatchSD batchSD(hypnos, 16); //set batch size uploading

int packetNumber = 0;
void setup()
{
    // Start the serial interface
    manager.beginSerial();

    // Enable the power rails on the hypnos
    hypnos.enable();

    setRTC();

    // Sets the LTE board to use batch SD to only start when we actually need to publish data
    lte.setBatchSD(batchSD);

    // load MQTT credentials from the SD card, if they exist
    mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

    // Initialize the modules
    manager.initialize();
}

void loop()
{
    // Wait 5 seconds for a message
    do{
[... 376 lines cut ...]
void setRTC(bool wait)
{
    if (!wait || !Serial)
        return;

    Serial.println(F("<Dendrometer> Adjust RTC time? (y/n)"));
    while (!Serial.available())
        ;
    int val = Serial.read();
    delay(50);
    while (Serial.available())
        Serial.read(); // flush the input buffer to avoid invalid input to rtc function

    if (val == 'y')
    {
        hypnos.set_custom_time();
    }
}
```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`

**Additional context**
This is different than the previous null packet issue

WL-Richards, 2024-03-06:
Can you paste your code for how you are measuring your magnet sensor here too

WL-Richards, 2024-03-07:
Can you upload a screen shot of the full packet in question on both the SD card and on MongoDB

nicholasqle, 2024-03-07:
For the spiky graph above, the packet that shows in MongoDB has a slightly different time and dramatically different displacement value (second to last column. 
[image]

[image]

As for the undefined null field, I do not have that exact one from SD card but here is two more I found. The undefined will occasionally appear in different fields. Sometimes the values are incorrect but in this case it is close
[image]

[image]

WL-Richards, 2024-03-08:
I'd recommend turning on SD_DEBUG logging if it's not already on. https://github.com/OPEnSLab-OSU/Loom-V4/wiki/Utilizing-the-Loom-Logger There is the link to how to enable it, if it is already on could you send me the file that has the same packet numbers as the broken ones I wanna see whats happening on the device

nicholasqle, 2024-04-17:
I set up multiple devices to upload at the same time, and it does look like packet from multiple devices at the same time are causing the problems. 
SD Debug from LTE hub:
```
[2024.04.14 17:49:57] [DEBUG] [Loom_LoRa.cpp:receiveBatch:340] Received packet!
[2024.04.14 17:49:57] [DEBUG] [Loom_Manager.cpp:display_data:198] Data Json: 

{
  "type": "data",
  "id": {
    "name": "DendrometerV4_",
    "instance": 29
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 411
      }
    },
    {
      "type": null,
      "id": {
        "name": null,
        "instance": 0
      },
      "numPackets": 0,
      "contents": []
    },
    {
      "type": "data",
      "id": {
        "name": "DendrometerV4_",
        "instance": 27
      },
      "numPackets": 7,
      "contents": [],
      "timestamp": {
        "time_utc": "2024-04-14T22:4:50Z",
        "time_local": "2024-04-14T15:4:50Z"
      }
    },
    {
      "module": "Packet",
      "data": {
        "Number": 401
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.165283203,
        "Vbat_MV": 4162.060547
      }
    },
    {
      "module": "SHT31",
      "data": {
        "Temperature": 20.12000084,
        "Humidity": 52.11000061,
        "VPD": 1.128057599
      }
    },
    {
      "module": "LoRa",
      "data": {
        "RSSI": -79
      }
    }
  ],
  "timestamp": {
    "time_utc": "2024-04-15T0:33:24Z",
    "time_local": "2024-04-14T17:33:24Z"
  }
}
[2024.04.14 17:49:57] [DEBUG] [SDManager.cpp:log:159] Successfully logged data to Hub2.csv
[2024.04.14 17:49:57] [DEBUG] [Loom_MongoDB.cpp:publish:148] Batch not ready to publish: 10/16
[2024.04.14 17:49:57] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:00] [DEBUG] [Loom_LoRa.cpp:receiveBatch:312] Packet Received!
[2024.04.14 17:50:00] [ERROR] [Loom_LoRa.cpp:receiveBatch:336] Failed to receive packet!
[2024.04.14 17:50:00] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:01] [DEBUG] [Loom_LoRa.cpp:receiveBatch:312] Packet Received!
[2024.04.14 17:50:01] [ERROR] [Loom_LoRa.cpp:receiveBatch:336] Failed to receive packet!
[2024.04.14 17:50:01] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:03] [DEBUG] [Loom_LoRa.cpp:receiveBatch:312] Packet Received!
[2024.04.14 17:50:03] [ERROR] [Loom_LoRa.cpp:receiveBatch:336] Failed to receive packet!
[2024.04.14 17:50:03] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:05] [DEBUG] [Loom_LoRa.cpp:receiveBatch:312] Packet Received!
[2024.04.14 17:50:05] [DEBUG] [Loom_LoRa.cpp:receivePartial:372] Waiting for packet 1 / 7
[2024.04.14 17:50:06] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:06] [DEBUG] [Loom_LoRa.cpp:receivePartial:377] Fragment received 1 / 7
[2024.04.14 17:50:07] [DEBUG] [Loom_LoRa.cpp:receivePartial:372] Waiting for packet 2 / 7
[2024.04.14 17:50:07] [DEBUG] [Loom_LoRa.cpp:recv:146] Waiting for packet...
[2024.04.14 17:50:08] [DEBUG] [Loom_LoRa.cpp:receivePartial:377] Fragment received 2 / 7
[2024.04.14 17:50:08] [DEBUG] [Loom_LoRa.cpp:receivePartial:372] Wa
```

The full file is too big to upload but I think you can get the idea, I can send it if you need it.

nicholasqle, 2024-04-22:
Chet wanted to check in on this to make sure the packet gets pushed to the correct device

udellc, 2024-07-12:
@WL-Richards , can we look into LoRa Radiohead library channel activity functionality to see if we can avoid collisions?

sarveshta, 2025-01-21:
- Working on getting Loom to compile on new computer
- david's been working on prototype for Lora fix
- they have a flowchart from last term
- each device already has an id
- 
- should be done soon

- about to start working on mqtt - talked to will to ensure duplicity doesn't happen
- they have hardware to test Lora (x3 boards - two trans and one receiver)

udellc, 2025-02-14:
Looking to see if we made progress on this and a plan to close it out

rainewheary, 2025-04-18:
Fixed with #176

PR #176 (2025-03-11): LoRa bugfix
Fixes #138, by allowing LoRa to receive data from multiple transmitters simultaneously.
closed 2025-04-18 by rainewheary