OPEnSLab-OSU/Loom-V4 #140: Loom_WIFI:isConnected() and MQTTComponent:setMaxRetries are protected instead of public causing compiling error

nicholasqle, 2024-03-02:
**Describe the bug**
Certain functions in Loom that is used for the WiFi dendrometers have been set as protected instead of public in Loom 4.6

**Hardware in Use**
Feather M0 WiFi

**To Reproduce**
Steps to reproduce the behavior:
Compile dendrometer code with WiFi defined

**Expected behavior**
Should compile and work as with older versions of Loom.

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

```cpp
#include <Loom_Manager.h> //4.6
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Hardware/Actuators/Loom_Neopixel/Loom_Neopixel.h>
#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/Analog/Loom_Teros10/Loom_Teros10.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>
#include <Radio/Loom_LoRa/Loom_LoRa.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

#include "AS5311.h"

//////////////////////////
/* DEVICE CONFIGURATION */
//////////////////////////
static const uint8_t NODE_NUMBER = 2;
static const char * DEVICE_NAME = "BiosphereDendrometer_";
////Select one wireless communication option
//#define DENDROMETER_LORA
#define DENDROMETER_WIFI
////These two time values are added together to determine the measurement interval
static const int8_t MEASUREMENT_INTERVAL_MINUTES = 15;
static const int8_t MEASUREMENT_INTERVAL_SECONDS = 0;
static const uint8_t TRANSMIT_INTERVAL = 16; // to save power, only transmit every X measurements
////Use teros 10?
//#define DENDROMETER_TEROS10
//////////////////////////
//////////////////////////

// Pins
#define AS5311_CS A3 // 9 for LB version, A3 otherwise
#define AS5311_CLK A5
#define AS5311_DO A4
#define BUTTON_PIN A1 // 11 for LB, A1 otherwise

// Loom
Manager manager(DEVICE_NAME, NODE_NUMBER);
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST); // 3_2 for LB, 3_3 otherwise
// Loom Sensors
Loom_Analog analog(manager);
#ifdef DENDROMETER_TEROS10
Loom_Teros10 teros(manager, A0);
#endif
Loom_SHT31 sht(manager);
Loom_Neopixel statusLight(manager, false, false, true, NEO_RGB); // using channel 2 (physical pin A2). use RGB for through-hole LED devices. GRB otherwise.

// magnet sensor
AS5311 magnetSensor(AS5311_CS, AS5311_CLK, AS5311_DO);

// wireless
#if defined DENDROMETER_LORA && defined DENDROMETER_WIFI
#error Choose ONE wireless communication protocol.
#elif defined DENDROMETER_LORA
[... 275 lines cut ...]
    while (!Serial.available())
        ;
    int val = Serial.read();
    delay(50);
    while (Serial.available())
        Serial.read(); // flush the input buffer to avoid invalid input to rtc function

    if (val == 'y')
    {
        hypnos.set_custom_time();
    }
}
```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`
``````
Compilation error: 'virtual bool Loom_WIFI::isConnected()' is protected within this context
``````
``````
Compilation error: 'void MQTTComponent::setMaxRetries(int)' is protected within this context
``````
**Additional context**
Add any other context about the problem here.
I brought this up with Will who said to post this issue, and for now to move the methods I need to public as a temporary fix.

ZakaryW, 2024-03-06:
Here's a [branch](https://github.com/OPEnSLab-OSU/Loom-V4/commits/connection-patch/) with a fix

WL-Richards, 2024-03-06:
This PR has been merged into main.

closed 2024-03-06 by WL-Richards