OPEnSLab-OSU/Loom-V4 #141: CONNECTION_REFUSED to broker when using WiFi MongoDB batch uploading

nicholasqle, 2024-03-04:
**Describe the bug**
The device fails to connect to the broker every time it tries to upload batch over WiFI.

**Hardware in Use**
A list of all hardware currently in use
Feather M0 WiFi, Hypnos 3.3
**To Reproduce**
Steps to reproduce the behavior:
1. Use WiFiMongoDBBatch
2. Wait for batch upload

**Expected behavior**
Should connect and publish packets

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here
This is just the Loom WiFiMongoDBBatch example
```cpp
/**
 * This is an example use case for Loomified wifi and MQTT to log data remotely in BATCHES
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */
#include "arduino_secrets.h"

#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Device", 1);

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

Loom_WIFI wifi(manager, CommunicationMode::CLIENT, SECRET_SSID, SECRET_PASS);
Loom_MongoDB mqtt(manager, wifi.getClient(), SECRET_BROKER, SECRET_PORT, DATABASE, BROKER_USER, BROKER_PASS, PROJECT);

// Enables batch logging with a batch size of 15
Loom_BatchSD batchSD(hypnos, 15);

void setup() {

  // Start serial
  manager.beginSerial();

  // Enable the Hypnos
  hypnos.enable();

  // Set an instance of BatchSD on the wifi module
  wifi.setBatchSD(batchSD);

  // Initialize all modules
  manager.initialize();
}

void loop() {
  // Package data
  manager.package();

  manager.display_data();

  // Need to log to SD to store the batch data
  hypnos.logToSD();

  // Pass batch SD along to the MQTT module
  mqtt.publish(batchSD);

  // Wait 5 seconds
  manager.pause(5000);
}
```

**Output**
Copy and paste the serial output here if possible wrapped in \`\`\` \`\`\`
``````
[2024.03.03 17:37:48] [DEBUG] [Loom_Manager.cpp:display_data:198] Data Json: 

{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 15
      }
    },
    {
      "module": "WiFi",
      "data": {
        "SSID": "OSU_Unsecured",
        "RSSI": -68
      }
    }
  ],
  "timestamp": {
    "time_utc": "2024-03-04T1:37:48Z",
    "time_local": "2024-03-03T17:37:48Z"
  }
}
[2024.03.03 17:37:48] [DEBUG] [SDManager.cpp:log:159] Successfully logged data to Device2.csv
[2024.03.03 17:37:48] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.03.03 17:38:08] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.03.03 17:38:13] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.03.03 17:38:33] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.03.03 17:38:38] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.03.03 17:38:58] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.03.03 17:39:03] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.03.03 17:39:23] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.03.03 17:39:28] [ERROR] [MQTTComponent.cpp:connectToBroker:39] MQTT Retry limit exceeded!
``````
**Additional context**
Add any other context about the problem here.
I was first facing this issue when using the Dendrometer WiFi code so I tried the Loom example code as a sanity check and it is still not working. 
I am using the same exact broker, database, port, user, password, etc information in arduino_secrets.h that I am for LTE which works fine.

WL-Richards, 2024-03-05:
Could you try using OSU Access? and also is your feather registered with then network https://selfreg.nws.oregonstate.edu/device_list

nicholasqle, 2024-03-06:
Took more than 24 hours for registration to work for some reason.

closed 2024-03-06 by nicholasqle