OPEnSLab-OSU/Loom-V4 #149: Test Hypnos Power Rail Changes

ZakaryW, 2024-04-25:
To make the hypnos power rail control easier, an enum system was implemented. This should be tested with different configurations of enabling/ disabling the power rails. These changes are on the latest version of the wisp spring patch.

WL-Richards, 2024-04-26:
[image]
I don't think this will work in its current state, as pin 5 and 6 are configured as OUTPUTs I don't think we can read the values in

ZakaryW, 2024-04-29:
Updated

ZakaryW, 2024-05-05:
Did my own testing and the change appears to be working.

udellc, 2024-07-12:
@SorenEmmons , this one is on you to test to confirm Zak's changes. Please get advice from Zak for testing

Sarvesh-Thiruppathi, 2024-07-25:
SETNONE - Getting 3.3/5v 
SET5 - Only getting 3.3v

SET3 & SET3_5 cause board to get stuck at the following:
```[DEBUG] [:initializeRTC:216] Initializing DS3231....```

Sarvesh-Thiruppathi, 2024-07-26:
SET5 - 3.3v when awake, 5v when asleep
SETNONE  - 3.3/5v when awake, none when asleep
SET3 - Gets stuck at ``[DEBUG] [:initializeRTC:216] Initializing DS3231....``
SET3_5 - Gets stuck at ``[DEBUG] [:initializeRTC:216] Initializing DS3231....``

**Test Setup**: FeatherM0 and Hypnos only
Testing with wisp-spring-patch branch loaded

Test code:

```/**
 * This is an example use case for the Hypnos board's sleep functionality
 * This allows the user to put the Feather into a deep sleep disabling power to all sensors and then resume operation after a given length of time
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */

#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

// Manager to control the device
Manager manager("Device", 1);

// Create a new Hypnos object setting the version to determine the SD Chip select pin, and starting without the SD card functionality
//Loom_Hypnos(Manager& man, HYPNOS_VERSION version, TIME_ZONE zone, bool use_custom_time = false, bool useSD = true)
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, false, false);

// Called when the interrupt is triggered 
void isrTrigger(){
  hypnos.wakeup();
}

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable(SET3_5);
  
  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);
}

void loop() {

  // Set the RTC interrupt alarm to wake the device in 10 seconds
  hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 40));

  // Reattach to the interrupt after we have set the alarm so we can have repeat triggers
  hypnos.reattachRTCInterrupt();
  
  delay(20000);
  // Put the device into a deep sleep, operation HALTS here until the interrupt is triggered
  hypnos.sleep();
}
```

Sarvesh-Thiruppathi, 2024-08-01:
After some debugging, I identified that the issue of the board getting hung at Initalizaing DS3231 occurs on the main branch as well when the following parameters are used:
```hypnos.enable(0,1);```

This issue occurs whenever initializeRTC is called after setting the pin5 to HIGH.

The traceback of the issue is as follows:

-> Loom_Hypnos.enable()
-> Loom_Hypnos.initializeRTC()
-> RTC_DS3231.lostPower()
-> OPEnS_RTC.cpp -> read_i2c_register()
-> Wire.endTransmission()
->  Somewhere in  Wire.endTransmission() -> ``if ( !sercom->startTransmissionWIRE( txAddress, WIRE_WRITE_FLAG ) )``

@WL-Richards

Sarvesh-Thiruppathi, 2024-08-01:
Additionally, when initalizeRTC is called before setting pin5 to HIGH, it doesn't get stuck at initializing DS3231. However, instead, it gets stuck at Registering interrupt:

```16:45:38.002 -> [DEBUG] [:registerInterrupt:149] Registering interrupt...

16:45:38.040 -> before pin > PINS_COUNT

16:45:38.040 -> after pin > PINS_COUNT

16:45:38.040 -> Before EEXt Interrupts

16:45:38.040 -> After EEXt Interrupts

16:45:38.040 -> Before attachInterrupt```

sarveshta, 2024-10-02:
Will re-test this soon. It is likely that I had a problem with my Loom installation that may have affected this testing based on my experience from debugging issue #153.

sarveshta, 2024-10-02:
Disregarding, the issue is still reoccurring. Arrived at this conclusion as SET3 & SET3_5 still get stuck at ``[DEBUG] [:initializeRTC:216] Initializing DS3231....``

ZakaryW, 2024-10-06:
Try testing 88bf89d0ed9088c777edb515356e9af3aa9ac5be to see if that resolved the issue.

sarveshta, 2024-10-11:
@sarveshta Test again on wisp-patch branch. Changes were made

sarveshta, 2024-10-18:
Test Results @ZakaryW 

**SETNONE**
When ON: Both on
When OFF: Both off

**SET3**
When ON: 5V on, 3V off (BUT gets stuck at ``[DEBUG] [:initializeRTC:207] Initializing DS3231….``)
When OFF: Never gets to this state

**SET5**
When ON: 3V ON, 5V OFF
When OFF: 5V OFF, 3V ON (I assume this is unintentional behavior. This likely happens as the toggle_power_rails function is called prior to going to sleep, thereby inverting the voltage configs.

**SET3_5**
When ON: Both turned off (BUT gets stuck at ``[DEBUG] [:initializeRTC:207] Initializing DS3231….``)
When OFF: Never gets to this stage

Tested with following code:
```
#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

Manager manager("Device", 1);

// Create a new Hypnos object setting the version to determine the SD Chip select pin, and starting without the SD card functionality
//Loom_Hypnos(Manager& man, HYPNOS_VERSION version, TIME_ZONE zone, bool use_custom_time = false, bool useSD = true)
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

// Called when the interrupt is triggered 
void isrTrigger(){
  hypnos.wakeup();
}

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable(SET3_5);

  // Called after enable
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);
}

void loop() {
  // Set the RTC interrupt alarm to wake the device in 10 seconds, done at top so evenly spaced sample periods 
  hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 20));

  // Measure and package data
  manager.measure();
  manager.package();
  
  // Print the current JSON packet
  manager.display_data();            

  // Log the data to the SD card              
  hypnos.logToSD();

  

  // Reattach to the interrupt after we have set the alarm so we can have repeat triggers
  hypnos.reattachRTCInterrupt();
  
  // Put the device into a deep sleep, operation HALTS here until the interrupt is triggered
  delay(10000);
  hypnos.sleep();
}
```

ZakaryW, 2024-10-20:
Updated and tested with the following code. Seems to be working for me.

```
#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

Manager manager("Device", 1);

volatile uint8_t itr = 0;
bool should3 = true;
bool should5 = true;

// Create a new Hypnos object setting the version to determine the SD Chip select pin, and starting without the SD card functionality
//Loom_Hypnos(Manager& man, HYPNOS_VERSION version, TIME_ZONE zone, bool use_custom_time = false, bool useSD = true)
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

// Called when the interrupt is triggered 
void isrTrigger(){
  if (itr < 3) itr++;
  else         itr = 0;
  hypnos.wakeup();
}

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable(HypnosPowerConfig::SET3_5);
  Serial.println("Enabled hypnos: 3v and 5v rails");

  // Called after enable
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  itr = 0;
}

void loop() {

  if (itr == 0) {
    should3 = true;
    should5 = true;
  }
  else if (itr == 1) {
    should3 = true;
    should5 = false;
  }
  else {
    should3 = false;
    should5 = true;
  }

  // Measure and package data
  manager.measure();
  manager.package();
  
  // Log the data to the SD card              
  hypnos.logToSD();

  // Print the current JSON packet
  manager.display_data();            
  // Both calls to this should print that they're on, sanity check
  hypnos.print_rail_info();
  // Testing functionality
  hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 30));
  hypnos.reattachRTCInterrupt();
  hypnos.sleep(true, should3, should5);
  // To check the power of the rails during sleep, look at the LED indicators
  hypnos.print_rail_info();
  delay(10000);
}
```

sarveshta, 2024-10-26:
### Test Report:

**SETNONE**
When ON: 3V ON, 5V OFF (but both are turned on the second cycle and thereafter)
When OFF: Both off

**SET3**
When ON: 3V ON, 5V OFF (but both are turned on the second cycle and thereafter)
When OFF: Both are turned off

**SET5**
When ON: 3V turns on, and then a couple of seconds later, 5V turns on, too (both continue to be turned on simultaneously second cycle and thereafter)
When OFF: Both turned off

**SET3_5**
When ON: Both turned on
When OFF: Both turned off

### Testing Code:
```
#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

Manager manager("Device", 1);

// Create a new Hypnos object setting the version to determine the SD Chip select pin, and starting without the SD card functionality
//Loom_Hypnos(Manager& man, HYPNOS_VERSION version, TIME_ZONE zone, bool use_custom_time = false, bool useSD = true)
Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

// Called when the interrupt is triggered 
void isrTrigger(){
  hypnos.wakeup();
}

void setup() {

  // Start the serial interface
  manager.beginSerial();

  // Enable the hypnos rails
  hypnos.enable(HypnosPowerConfig::SETNONE);

  // Called after enable
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);
}

void loop() {
  // Set the RTC interrupt alarm to wake the device in 10 seconds, done at top so evenly spaced sample periods 
  hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 20));

  // Measure and package data
  manager.measure();
  manager.package();
  
  // Print the current JSON packet
  manager.display_data();            

  // Log the data to the SD card              
  hypnos.logToSD();

  

  // Reattach to the interrupt after we have set the alarm so we can have repeat triggers
  hypnos.reattachRTCInterrupt();
  
  // Put the device into a deep sleep, operation HALTS here until the interrupt is triggered
  delay(10000);
  hypnos.sleep();
}
```

@ZakaryW @WL-Richards

WL-Richards, 2024-11-08:
Power rail changes have been rewriten and are merged into 4.7-merge

closed 2024-11-08 by WL-Richards