OPEnSLab-OSU/Loom-V4 #150: Figure out MS5803 initialization weirdness

WL-Richards, 2024-05-03:


udellc, 2024-09-11:
We neet to re-test after we updated the library to see if this is still an issue.

udellc, 2024-09-11:
Let's stablize this so we can rely on the CRC checks!

mbelinga777, 2024-09-11:
Hi @udellc 
After updating the library like will suggested, I conducted tests on five MS5803 sensors. Four of the sensors passed the CRC checks consistently, but one failed despite the data appearing accurate. It seems the issue with the failing sensor could be hardware-related.
It looks like the library update has improved overall stability regarding the CRC checks, but the single failing sensor may need further investigation.
Let me know if additional testing or troubleshooting is required.

udellc, 2024-09-11:
Thanks for the update @mbelinga777 !
@WL-Richards , can we be sure the updated MS5803 library is pushed to the main branch? Then close the issue?

WL-Richards, 2024-09-15:
Since this a non-Loom library it doesn't actually get pushed to main it just gets updated in the board profile

natashazai, 2024-10-02:
After updating the MS5803 library and uncommenting the code for MS5803.cpp, it initializes both osmopods now. Data is being read properly.
[image]

natashazai, 2024-10-02:
When there is only one osmopod connected, so the one that is on the board, it properly fails to initialize one of them as it should be
[image]

natashazai, 2024-10-02:
All osmopods were tested

closed 2024-10-02 by natashazai