OPEnSLab-OSU/Loom-V4 #152: Timestamp Lora is null

natashazai, 2024-06-15:
**Describe the bug**
The Lora hub is writing to the packet timestamp with both UTC and Local time, however, the recipient Lora has timestamp utc null.

**Hardware in Use**
A list of all hardware currently in use:
-Lora feather to transmit
-Lora feather to receive
-stemma sensor
-Hypnos connected to transmitting Lora
and stemma sensor
-3.8V battery

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Arduino IDE and add hypnos and stemma sensor library and object to the Lora single batch transmit code.
2. Upload the completed code to Lora feather, and upload the Lora receive code to the second feather.
3. Wire Transmitting Lora with hypnos, battery, and stemma sensor.
4. See the timestamp on the receiving Lora.

**Expected behavior**
The timestamp should look like this on Lora:
```
"timestamp": {
    "time_utc": "2024-06-14T15:55:32Z",
    "time_local": "2024-06-14T8:55:32Z"
  }
```

**Code**
Loom/Radio/Lora/Single Batch/Receive. ino code
Loom/Radio/Lora/Single Batch/Transmit. ino code + hypnos + stemma.
The code that I used:
```
/**
 * This is an example use case for LoRa communication
 * 
 * MANAGER MUST BE INCLUDED FIRST IN ALL CODE
 */
#include <Loom_Manager.h>

#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>
//#include <Hardware/Loom_BatchSD/Loom_BatchSD.h>
#include <Radio/Loom_LoRa/Loom_LoRa.h>
#include <Sensors/I2C/Loom_STEMMA/Loom_STEMMA.h>

Manager manager("Device", 1);

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST);

// Do we want to use the instance number as the LoRa address
Loom_LoRa lora(manager);
Loom_STEMMA stemma(manager);

// Create a batch size of 15
//Loom_BatchSD batch(hypnos, 2);

void isrTrigger(){
  hypnos.wakeup();
}

void setup() {
  manager.beginSerial();

  // Set a reference to the batchSD object
  //lora.setBatchSD(batch);

  hypnos.enable();

  manager.initialize();

  hypnos.registerInterrupt(isrTrigger);
}

void loop() {
  hypnos.setInterruptDuration(TimeSpan(0, 0, 0, 30));
  manager.measure();
  manager.package();
  manager.display_data();

  // Log the hypnos to the SD card
  //hypnos.logToSD();

  // Send the current JSON document to address 0
  //lora.sendBatch(0);
  lora.send(0);

  // Wait 5 seconds between transmits
  //manager.pause(5000);

   hypnos.reattachRTCInterrupt();

     hypnos.sleep();
}
```

**Output**
The timestamp on the receiving lora looks null:
```
"timestamp": {
    "time_utc": null
  }
```

**Additional context**
Add any other context about the problem here.

WL-Richards, 2024-10-11:
Do you still the logs from this?

sarveshta, 2025-02-11:
Quinn was using Lora for his onboarding exam. I will get in touch to see if he's having any issues similar to this.

sarveshta, 2025-02-18:
The issue cannot be reproduced.

closed 2025-02-18 by sarveshta