OPEnSLab-OSU/Loom-V4 #153: Combine reading time zone and sample interval from SD into one function

CaseyJ37, 2024-07-05:
Make one function that reads the time zone and the sleep interval from the SD card.

WL-Richards, 2024-09-23:
This has been implemented and tested on my machine and it appears to work, still need testing on one additional machine

sarveshta, 2024-10-02:
Works! 

Tested with ``Hypnos_SD_Sleep_Interval_From_SD`` example and worked successfully. Serial monitor output attached below for reference:

```
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:set_custom_time:358] Custom time successfully set to: 2024.11.02 00:09:23
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:initializeRTC:198] DS3231 Real-Time Clock Initialized Successfully!
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:initializeRTC:201] Custom time successfully set to: 2024.11.02 00:09:23
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:getConfigFromSD:489] Config successfully loaded from SD!
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:getConfigFromSD:493] Selected timezone: AKST, UTC offset: -9
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:registerInterrupt:104] Registering interrupt...
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:registerInterrupt:116] Interrupt successfully attached!
17:10:04.666 -> [2024.11.02 00:09:23] [DEBUG] [:setInterruptDuration:374] Current Time (Local): 2024.11.01 15:09:23
17:10:04.700 -> [2024.11.02 00:09:23] [DEBUG] [:setInterruptDuration:377] Next Interrupt Alarm Set For: 2024.11.01 15:09:33
17:10:04.700 -> [2024.11.02 00:09:23] [DEBUG] [:display_data:202] JSON Document is Null there is no data to display
17:10:04.733 -> [2024.11.02 00:09:23] [DEBUG] [:log:159] Successfully logged data to Device2.csv
17:10:04.733 -> [2024.11.02 00:09:23] [DEBUG] [:reattachRTCInterrupt:157] Interrupt successfully reattached!
17:10:04.733 -> [2024.11.02 00:09:23] [DEBUG] [:sleep:396] Entering Standby Sleep...
```

closed 2024-10-02 by sarveshta