OPEnSLab-OSU/Loom-V4 #157: Test out i2c MUX implementation on Chimes

udellc, 2024-09-11:
We want to add the Loom i2c multiplexer to the hardware to solve i2c bus issues and provide more flexibility for hot swapping sensors. Need to go back and test i2c MUX with Chimes sensors.

SorenEmmons, 2024-10-03:
Tested TCA8548A Multiplexer using the multiplexer example, tested using one and two TSL2591 on both tests the mux was able to detect the two sensors on all i2c and output data and were responsive to changes in light. The only thing of note was that on all of the channels including the ones where a tsl2591s was connected to the multiplexer read that there was an MB1232 on that channel for further detail look at the attached screenshot of the serial monitor output. 

[image]

WL-Richards, 2024-10-04:
What does the initialization procedure output at the start?

WL-Richards, 2024-10-04:
Ah I see, the mux's default I2C address is 0x70 which is also the address for an MB1232, you need to short the address pin for the mux to vin on any of the following addresses: 0x71, 0x72, 0x73, 0x74, 0x75, 0x78. This is a requirement for the multiplexer to work as these are the only possible addresses that don't conflict with existing I2C devices in Loom.

SorenEmmons, 2024-10-04:
Shorted the A0 pin to add one to the address so the new address is 0x71 and the multiplexer stopped detecting the MB1232 on all the channels and now works as expected.
[image]

closed 2024-10-11 by WL-Richards