OPEnSLab-OSU/Loom-V4 #161: LTE Connection Refused

ZakaryW, 2024-10-25:
When running the LTE changes on the whisp spring patch, the following occurs:
```
[SD Manager] Initializing SD Card...
[SD Manager] Successfully initialized SD Card!
[SD Manager] Data will be logged to Whisp_Proto3.csv
[DEBUG] [Loom_Hypnos.cpp:initializeRTC:224] Initializing DS3231....
[DEBUG] [Loom_Hypnos.cpp:initializeRTC:248] DS3231 Real-Time Clock Initialized Successfully!
[2024.10.25 22:38:15] [DEBUG] [Loom_Hypnos.cpp:initializeRTC:251] Custom time successfully set to: 2024.10.25 22:38:15
[2024.10.25 22:38:15] [DEBUG] [Loom_Manager.cpp:initialize:217] ** Initializing Modules **
[2024.10.25 22:38:15] [ERROR] [Loom_SEN55.cpp:initialize:34] Error occurred while attempting to reset device: Received NACK on transmit of address, module will not be initialized!
[2024.10.25 22:38:15] [ERROR] [Loom_SHT31.cpp:initialize:22] Failed to initialize SHT31! Check connections and try again...
[2024.10.25 22:38:15] [DEBUG] [Loom_LTE.cpp:power_up:139] Powering up GPRS Modem. This should take about 10 seconds...
[2024.10.25 22:38:27] [DEBUG] [Loom_LTE.cpp:power_up:149] Powering up complete!
[2024.10.25 22:38:28] [DEBUG] [Loom_LTE.cpp:connect:201] Waiting for network...
[2024.10.25 22:38:28] [DEBUG] [Loom_LTE.cpp:connect:214] Connected to network!
[2024.10.25 22:38:28] [DEBUG] [Loom_LTE.cpp:connect:218] Attempting to connect to LTE Network: hologram
[2024.10.25 22:38:29] [DEBUG] [Loom_LTE.cpp:connect:220] Successfully Connected!
[2024.10.25 22:38:41] [DEBUG] [Loom_LTE.cpp:initialize:95] Connected!
[2024.10.25 22:38:41] [DEBUG] [Loom_LTE.cpp:initialize:99] APN: hologram
[2024.10.25 22:38:41] [DEBUG] [Loom_LTE.cpp:initialize:103] Signal State: 26
[2024.10.25 22:38:41] [DEBUG] [Loom_LTE.cpp:initialize:108] Device IP Address: 100.75.67.65
[2024.10.25 22:38:41] [DEBUG] [Loom_LTE.cpp:initialize:111] Module successfully initialized!
[2024.10.25 22:38:41] [DEBUG] [Loom_Manager.cpp:initialize:223] ** Setup Complete ** 
[2024.10.25 22:38:41] [DEBUG] [Loom_Hypnos.cpp:registerInterrupt:154] Registering interrupt...
[2024.10.25 22:38:41] [DEBUG] [Loom_Hypnos.cpp:registerInterrupt:166] Interrupt successfully attached!
[2024.10.25 22:38:41] [DEBUG] [Loom_Manager.cpp:measure:68] ** Measuring **
[2024.10.25 22:38:41] [WARNING] [Loom_Manager.cpp:measure:77] SEN55 Not initialized!
[2024.10.25 22:38:41] [WARNING] [Loom_Manager.cpp:measure:77] SHT31 Not initialized!
[2024.10.25 22:38:41] [DEBUG] [Loom_Manager.cpp:measure:85] ** Measuring Complete **
[2024.10.25 22:38:41] [DEBUG] [Loom_Manager.cpp:package:95] ** Packaging **
[2024.10.25 22:38:41] [WARNING] [Loom_Manager.cpp:package:119] SEN55 Not initialized!
[2024.10.25 22:38:41] [WARNING] [Loom_Manager.cpp:package:119] SHT31 Not initialized!
[2024.10.25 22:38:41] [DEBUG] [Loom_Manager.cpp:package:125] ** Packaging Complete **
```
Then when ready to send the batch...
```
      "module": "Analog",
      "data": {
        "Vbat": 4.315136909,
        "Vbat_MV": 4305.46875
      }
    },
    {
      "module": "LTE",
      "data": {
        "RSSI": 99
      }
    }
  ],
  "timestamp": {
    "time_utc": "2024-10-25T22:41:58Z",
    "time_local": "2024-10-25T15:41:58Z"
  }
}
[2024.10.25 22:42:00] [DEBUG] [SDManager.cpp:log:159] Successfully logged data to Whisp_Proto3.csv
[2024.10.25 22:42:02] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.10.25 22:42:05] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.10.25 22:42:12] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.10.25 22:42:15] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.10.25 22:42:22] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.10.25 22:42:25] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.10.25 22:42:32] [DEBUG] [MQTTComponent.cpp:connectToBroker:46] Attempting to connect to broker: cas-mosquitto.biossys.oregonstate.edu:1883
[2024.10.25 22:42:35] [ERROR] [MQTTComponent.cpp:connectToBroker:51] Failed to connect to broker: CONNECTION_REFUSED
[2024.10.25 22:42:42] [ERROR] [MQTTComponent.cpp:connectToBroker:39] MQTT Retry limit exceeded!
[2024.10.25 22:42:42] [DEBUG] [Loom_Hypnos.cpp:setInterruptDuration:422] Current Time (Local): 2024.10.25 15:42:42
[2024.10.25 22:42:42] [DEBUG] [Loom_Hypnos.cpp:setInterruptDuration:425] Next Interrupt Alarm Set For: 2024.10.25 15:43:42
[2024.10.25 22:42:42] [DEBUG] [Loom_Hypnos.cpp:reattachRTCInterrupt:207] Interrupt successfully reattached!
[2024.10.25 22:42:42] [WARNING] [Loom_Manager.cpp:power_down:178] SEN55 Not initialized!
[2024.10.25 22:42:42] [WARNING] [Loom_Manager.cpp:power_down:178] SHT31 Not initialized!
[2024.10.25 22:42:42] [DEBUG] [Loom_LTE.cpp:power_down:170] Powering down GPRS Modem. This should take about 5 seconds...
```
Not entirely sure the cause, it should be noted the recent hypnos changes are also on this branch. Let me know if more info is needed.

WL-Richards, 2024-11-01:
Is that behavior strictly limited to that branch?

WL-Richards, 2024-11-08:
It appeared to be an issue with howe the new power on/ power off code was handling the lte board, this has been fixed in 4.7-merge. Leaving open until confirmed stable

closed 2024-11-08 by WL-Richards