OPEnSLab-OSU/Loom-V4 #162: SD Card Becoming Corrupted

SorenEmmons, 2024-10-27:
**Describe the bug**
Debug File and other CSV files are being corrupted causing the Hypnos not to be able to log data on 2 Wisp devices in the field. 

**Hardware in Use**

- Feather Mo Wifi 
- Hypnos 3.3 
- Sparkfun LTE board
- SEN55
- SHT31

**To Reproduce**
Steps to reproduce the behavior:
Finding an exact cause of why this behavior occurred has the debug file become corrupted so we can not look at the logs. But this unit was running the Wisp Batch logging code and leaving the unit to sit for about a month, though this issue did not occur in any other units to my knowledge in similar circumstances in the field.

**Expected behavior**
A clear and concise description of what you expected to happen.
The SD card will become corrupted with the debug and some of the other files being corrupted.

**Code**

```cpp
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Whisp_brd_v0p4_", 55); //change!!!!!!

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MongoDB mqtt(manager, lte.getClient());
//A batch is logged every 5 minutes, so 12 per hour (12 * 6 = 72) so mqtt will publish at batch size of 72/ every 6 hours
Loom_BatchSD batchSD(hypnos, 5);

void isrTrigger()
{
  hypnos.wakeup();
}

void setup() {
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Set the LTE board to only powerup when a batch is ready to be sent
  lte.setBatchSD(batchSD);

  // Enable the hypnos rails
  hypnos.enable();

  //Time Sync Using LTE 
  hypnos.setNetworkInterface(&lte);

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  hypnos.networkTimeUpdate();

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  // Log the data to the SD
  hypnos.logToSD();

  // Pass in the batchSD to the mqtt obj to check/ publish a batch of data if ready
  mqtt.publish(batchSD);
 
  // Set the interrupt duration for 5 minutes
  hypnos.setInterruptDuration(TimeSpan(0,0,0,5));

  // Reattach the interrupt
  hypnos.reattachRTCInterrupt();
  
  // Set the hypnos to sleep, but with power still being supplied to the 5v rail (wait for serial when testing from a computer)
  hypnos.sleep(false, true, false);

  //hypnos.networkTimeUpdate();
  //manager.pause(1000);
}

```

**Output**
[image]

**Additional context**
This happened in the field with the 4G board possibly being broken at this point as this happened towards the end of the season.

WL-Richards, 2024-11-01:
Chet suggested wiping/swapping the SD card and running it again to see if the device persists

SorenEmmons, 2024-11-06:
I already did that, all that was needed was to delete the debug file because when that fails the SD card refuses to initialize.

nicholasqle, 2024-11-22:
I have had a similar issue where we no longer had logged data after a certain date where a corrupted file was produced on the SD card (LBD~C36F.C[V). The SD card will fail to log with the corrupted file on it.
[image]
```
[2024.11.15 11:32:31]  [DEBUG]  [Loom_Manager.cpp:package:125] ** Packaging Complete **
[SD Manager] Failed to open log file!
```
I did not have any debug files. I ended up swapping for a new SD card to make sure it works again.

closed 2025-01-08 by sarveshta