OPEnSLab-OSU/Loom-V4 #167: Metadata Example

WL-Richards, 2024-12-06:
For writing metadata to the SD card for each device it should be written in the following format:
`{
    "type": "metadata",
    "latitude": "",
    "longitude": "",
    "owner": "",
    "deployment_date": "",
    "notes": "",
    "keywords": [
        "",
        "",
        ""
    ]
}`

Furthermore this should just be uploaded as the first packet (before data is transmitted) so that the packet shows up first in the given collection for each devices

sarveshta, 2025-01-14:
could make a function like ``mqtt.publishMeta(hypnos.readFile("metadata.json"));``
Using the same function definition as ``mqtt.publish()`` but publishes the metadata file instead of ``manInst->getJSONString(jsonString);``

sarveshta, 2025-01-24:
I'm assigning a review of this to @quinnedgar. You can find my code on the metadata branch.

I have included an example under Loom->Hypnos-> Hypnos_SD_Metadata. Try uploading it to a device and using Wifi/LTE and see if you can get a metadata packet uploaded to mongoDB. Contact me if you have any questions.

quinnedgar, 2025-02-07:
Tested upload on Chime3 and MongoDB logging to WhaleFest under Device1. Upload successful. 

Done with Sarvesh as a guide.

closed 2025-02-11 by sarveshta