OPEnSLab-OSU/Loom-V4 #168: Wisp Units Crashing

sarveshta, 2025-01-24:
Wisp units have been crashing during batch send.

Wisp 52
- x1 crashed halfway through (RSSI at -99dB, fully plugged in)
- The same chime failed with RSSI -22 and fully plugged in (20 days & connection refused on 576, reconnected next upload, connection refused on 4392, never reconnected, and finally crashed on 5401). 
The hologram says it's still connected for multiple days.
- 

Wisp 51
- x1 Wisp died during packet #40/72 (battery voltage at 3.64)

Soren speculates its 4G related or with Batch. 
Soren will send Wisp 52 and 51 file.

sarveshta, 2025-02-11:
## Wisp 54
```
[2025.01.18 14:16:31] [DEBUG] [Loom_MongoDB.cpp:publish:148] Batch not ready to publish: 71/72
[2025.01.18 14:16:31] [DEBUG] [Loom_Hypnos.cpp:setInterruptDuration:448] Current Time (Local): 2025.01.18 06:16:31
[2025.01.18 14:16:31] [DEBUG] [Loom_Hypnos.cpp:setInterruptDuration:451] Next Interrupt Alarm Set For: 2025.01.18 06:21:31
[2025.01.18 14:16:31] [DEBUG] [Loom_Hypnos.cpp:reattachRTCInterrupt:231] Interrupt successfully reattached!
[2025.01.18 14:16:31] [DEBUG] [Loom_Hypnos.cpp:sleep:470] Entering Standby Sleep...
[2025.01.18 14:21:32] [DEBUG] [Loom_LTE.cpp:power_up:134] Powering up GPRS Modem. This should take about 10 seconds...
[2025.01.18 14:21:44] [DEBUG] [Loom_LTE.cpp:power_up:144] Powering up complete!
[2025.01.18 14:21:44] [DEBUG] [Loom_LTE.cpp:connect:196] Waiting for network...
[2025.01.18 14:21:45] [DEBUG] [Loom_LTE.cpp:connect:209] Connected to network!
[2025.01.18 14:21:45] [DEBUG] [Loom_LTE.cpp:connect:213] Attempting to connect to LTE Network: hologram
[2025.01.18 14:21:45] [DEBUG] [Loom_LTE.cpp:connect:215] Successfully Connected!
[2025.01.18 14:21:51] [DEBUG] [Loom_Hypnos.cpp:post_sleep:550] Device has awoken from sleep!
[2025.01.18 14:21:51] [DEBUG] [Loom_Hypnos.cpp:networkTimeUpdate:329] Attempting to set RTC time to the current network time...
[2025.01.18 14:21:52] [DEBUG] [Loom_Hypnos.cpp:networkTimeUpdate:335] Network time successfully set to: 2025.01.18 14:21:52
[2025.01.18 14:21:52] [DEBUG] [Loom_Manager.cpp:measure:68] ** Measuring **
[2025.01.18 14:21:52] [DEBUG] [Loom_SEN55.cpp:measure:95] Waiting for data to be ready... If not ready in 10 seconds we will stop trying
[2025.01.18 14:21:52] [ERROR] [Loom_SEN55.cpp:measure:124] No new data was ready within the given time period.
[2025.01.18 14:21:52] [DEBUG] [Loom_Manager.cpp:measure:85] ** Measuring Complete **
[2025.01.18 14:21:52] [DEBUG] [Loom_Manager.cpp:package:95] ** Packaging **
[2025.01.18 14:21:52] [DEBUG] [Loom_Manager.cpp:package:125] ** Packaging Complete **
[2025.01.18 14:21:52] [DEBUG] [Loom_Manager.cpp:display_data:198] Data Json: 

{
  "type": "data",
  "id": {
    "name": "Whisp_brd_v0p4_",
    "instance": 52
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1584
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.365087986,
        "Vbat_MV": 4439.208984
      }
    },
    {
      "module": "SEN55",
      "data": {
        "PM1_0": 4.400000095,
        "PM2_5": 2.700000048,
        "PM4_0": 3.900000095,
        "PM10_0": 4.400000095,
        "AmbientHumidity": 41.84000015,
        "AmbientTemperature": 21.79999924,
        "VocIndex": 106,
        "NoxIndex": 1
      }
    },
    {
      "module": "SHT31",
      "data": {
        "Temperature": 20.56999969,
        "Humidity": 34.27999878
      }
    },
    {
      "module": "LTE",
      "data": {
        "RSSI": 24
      }
    }
[... 71 lines cut ...]
[2025.01.18 14:23:34] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 32 of 72
[2025.01.18 14:23:36] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:37] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 33 of 72
[2025.01.18 14:23:39] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:40] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 34 of 72
[2025.01.18 14:23:42] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:43] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 35 of 72
[2025.01.18 14:23:45] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:46] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 36 of 72
[2025.01.18 14:23:48] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:49] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 37 of 72
[2025.01.18 14:23:52] [DEBUG] [MQTTComponent.cpp:publishMessage:99] Data has been successfully sent!
[2025.01.18 14:23:52] [DEBUG] [Loom_MongoDB.cpp:publish:111] Publishing Packet 38 of 72
```

I was looking at the last couple packets, found it interesting that RSSI was 99 for a good portion of the packets except the one right before the batch send:
```
{"type":"data","id":{"name":"Whisp_brd_v0p4_","instance":52},"contents":[{"module":"Packet","data":{"Number":1579}},{"module":"Analog","data":{"Vbat":4.329638481,"Vbat_MV":4353.808594}},{"module":"SEN55","data":{"PM1_0":4.400000095,"PM2_5":2.700000048,"PM4_0":3.900000095,"PM10_0":4.400000095,"AmbientHumidity":41.84000015,"AmbientTemperature":21.79999924,"VocIndex":106,"NoxIndex":1}},{"module":"SHT31","data":{"Temperature":20.54999924,"Humidity":34.27999878}},{"module":"LTE","data":{"RSSI":99}}],"timestamp":{"time_utc":"2025-01-18T13:56:18Z","time_local":"2025-01-18T5:56:18Z"}}
{"type":"data","id":{"name":"Whisp_brd_v0p4_","instance":52},"contents":[{"module":"Packet","data":{"Number":1580}},{"module":"Analog","data":{"Vbat":4.363476753,"Vbat_MV":4363.476563}},{"module":"SEN55","data":{"PM1_0":4.400000095,"PM2_5":2.700000048,"PM4_0":3.900000095,"PM10_0":4.400000095,"AmbientHumidity":41.84000015,"AmbientTemperature":21.79999924,"VocIndex":106,"NoxIndex":1}},{"module":"SHT31","data":{"Temperature":20.53000069,"Humidity":34.33000183}},{"module":"LTE","data":{"RSSI":99}}],"timestamp":{"time_utc":"2025-01-18T14:1:21Z","time_local":"2025-01-18T6:1:21Z"}}
[... cut ...]
{"type":"data","id":{"name":"Whisp_brd_v0p4_","instance":52},"contents":[{"module":"Packet","data":{"Number":1584}},{"module":"Analog","data":{"Vbat":4.365087986,"Vbat_MV":4439.208984}},{"module":"SEN55","data":{"PM1_0":4.400000095,"PM2_5":2.700000048,"PM4_0":3.900000095,"PM10_0":4.400000095,"AmbientHumidity":41.84000015,"AmbientTemperature":21.79999924,"VocIndex":106,"NoxIndex":1}},{"module":"SHT31","data":{"Temperature":20.56999969,"Humidity":34.27999878}},{"module":"LTE","data":{"RSSI":24}}],"timestamp":{"time_utc":"2025-01-18T14:21:52Z","time_local":"2025-01-18T6:21:52Z"}}
```
I wonder if the batch send could've failed due to losing LTE connection.

rainewheary, 2025-04-18:
@SorenEmmons is this still an issue? If not let's close this.