OPEnSLab-OSU/Loom-V4 #181: Zak/Wisp MultiGas Integration

sarveshta, 2025-02-25:
Zak has integrated the DFRobot Multigas sensor into the Loom library. Branch DF_Sensors. Sarvesh is testing.

rainewheary, 2025-04-11:
@ZakaryW @SorenEmmons can you update on what the status is with this? I heard from Soren that he was validating this and was having issues. LMK what's up and I can complete the validation.

WL-Richards, 2025-04-11:
`
        const std::vector<std::string> gasTypes = {
            "O2", "CO", "H2S", "NO2", "O3", "CL2",
            "NH3", "H2", "HCL", "SO2", "HF", "PH3"
        };
`

This call in measure every time always allocates new memory possible heap fragmentation problem

ZakaryW, 2025-04-13:
See latest commit

rainewheary, 2025-04-15:
Hey Zak/Will, where are you pulling the DFRobot_MultiGasSensor dependency from?

WL-Richards, 2025-05-16:
I have rewritten the implementation of this and pushed it to DF_Sensors (pending validation)

I added a custom function to the lib (returns const char* instead of String (yikes))
Below for now is the zipped modified library

[DFRobot_MultiGasSensor-opens.zip](https://github.com/user-attachments/files/20258300/DFRobot_MultiGasSensor-opens.zip)

closed 2025-06-16 by WL-Richards