OPEnSLab-OSU/Loom-V4 #184: Data not appearing on MongoDB

SorenEmmons, 2025-03-09:
**Describe the bug A clear and concise description of what the bug is.**

Wisp units not uploading data to mongoDB, but according to the serial output and hologram, the units are sending successful packets to the server, it appears that all have happened on similar days around 02/06/25 - 02/07/25 occurs with all current wisp units. 

**Hardware in Use A list of all hardware currently in use**

Feather M0
Hypnos
Sparkfun LTE Board 
SHT31 
SEN55

**To Reproduce Steps to reproduce the behavior:**
Using the hardware used above and using the code attached and this as the mqtt creds run the device for about 75mins so that batch is reached (you can lower this) and let the device upload to mongoDB

`{
    "broker": "cas-mosquitto.biossys.oregonstate.edu",
    "port": 1883,
    "username": "User",
    "password": "password",
    "database": "Wisp_LTE_test",
    "project": ""
}`

Expected behavior A clear and concise description of what you expected to happen.
The expected behavior is we should be seeing the data collected by the wisp sensors should be uploading to mongoDB.

Code In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

Output Copy and paste the serial output here if possible wrapped in ```
```cpp
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Whisp_brd_v0p4_",13); //change

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MongoDB mqtt(manager, lte);
//A batch is logged every 5 minutes, so 12 per hour (12 * 6 = 72) so mqtt will publish at batch size of 72/ every 6 hours
Loom_BatchSD batchSD(hypnos, 15);

void isrTrigger()
{
  hypnos.wakeup();
}

void setup() {
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Set the LTE board to only powerup when a batch is ready to be sent
  lte.setBatchSD(batchSD);

  // Both power rails should be on when awake
  hypnos.setWakeConfiguration(POWERRAIL_CONFIG::PR_3V_ON_5V_ON);

  // Only the 5V rail should be on during sleep
  hypnos.setSleepConfiguration(POWERRAIL_CONFIG::PR_3V_OFF_5V_ON);

  // Enable the hypnos rails
  hypnos.enable();

  //Time Sync Using LTE 
  hypnos.setNetworkInterface(&lte);

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  hypnos.networkTimeUpdate();

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  // Log the data to the SD
  hypnos.logToSD();

  // Pass in the batchSD to the mqtt obj to check/ publish a batch of data if ready
  mqtt.publish(batchSD);
  //mqtt.publish();

  // Set the interrupt duration for 5 minutes
  hypnos.setInterruptDuration(TimeSpan(0,0,5,0));

  // Reattach the interrupt
  hypnos.reattachRTCInterrupt();
  
  // Set the hypnos to sleep, but with power still being supplied to the 5v rail (wait for serial when testing from a computer)
  hypnos.sleep(false);

  hypnos.networkTimeUpdate();
  //manager.pause(2000);
} 
```
 ```

Additional context Add any other context about the problem here.

rainewheary, 2025-05-15:
Current status: It seems that this issue only occurs on Soren's computer?

WL-Richards, 2025-06-16:
Closing for now because spookiness

closed 2025-06-16 by WL-Richards