OPEnSLab-OSU/Loom-V4 #193: NetworkTimeUpdate function sets the current time on the device to the current UTC time.

CaseyJ37, 2025-05-09:
**Describe the bug**
NetworkTimeUpdate function sets the current time on the device to the current UTC time. 

**Hardware in Use**
Normal Chimes hardware.

**To Reproduce**
Steps to reproduce the behavior:
Run the normal Chimes code using either the "4.7-merge" branch or the "chimes-dev" branch.

rainewheary, 2025-05-15:
Is this an issue you're detecting from logs, or have you been able to reproduce it while connected to Serial?

WL-Richards, 2025-06-16:
Upon looking into this the data we are receiving from the cell towers seems to vary, sometimes it responds with UTC time and an offset to local and other times it appears to respond with the local time and the offset to UTC and there is seemingly no way of knowing which

udellc, 2025-11-14:
[SARA-R42-LEXI-R42-Application-Development_AppNote_UBX-20050829.pdf](https://github.com/user-attachments/files/23536791/SARA-R42-LEXI-R42-Application-Development_AppNote_UBX-20050829.pdf)