OPEnSLab-OSU/Loom-V4 #195: Local timestamp missing Z at end on MongoDB

udellc, 2025-05-13:
**Describe the bug**
Timestamps on MongoDB for local time are missing the Z character. This might not be an issue in the MongoDB plotter, but in my webapp plotter ear2earth, I'm getting "Invalid data" and interleaved data plots mixing a lot of different sessions on my web app plotter. 

Note, Data sets where Sarvesh took SD card data from Chimes, and ran it though his script to upload onto MongoDB display and work well. I'll post screen shots of the difference and give the database names.

Databases with issues: tested mostly the SpringBreakChimeTest database, any data set in there seems to have issues.

Databases that work: TEK Cascade > KutiChime13ISOTS
and WhaleFest23 > KhutiChime7

udellc, 2025-05-13:
[image]
[image]

sarveshta, 2025-05-19:
Prior versions of the ear2earth website required the timestamp to be in YYYY-MM-DDTHH:MM:SSZ format. Some timestamps on MongoDB were in the format YYYY-MM-DDTH:M:SSZ. Notice how it's H and M instead of HH and MM (two digits). Example:

**Problematic Format:** 
2025-04-12T1:8:13Z

**Working Format:**
2025-04-12T01:08:13Z

The ear2earth application was updated to handle both formats. Commit [667dc24](https://github.com/OPEnSLab-OSU/WeatherChimesWeb/commit/667dc24a1ee334fe2cf0272fef1477d899ca50a9)

This is how SpringBreakChimeTest.EZchime5 looks when loaded post-update. @udellc 

[image]

@WL-Richards, You're right that the local format does not require a Z suffix since it indicates Zulu (UTC) time. I've fixed my script. Thanks.

sarveshta, 2025-05-19:
> Prior versions of the ear2earth website required the timestamp to be in YYYY-MM-DDTHH:MM:SSZ format. Some timestamps on MongoDB were in the format YYYY-MM-DDTH:M:SSZ. Notice how it's H and M instead of HH and MM (two digits). Example:
> 
> **Problematic Format:** 2025-04-12T1:8:13Z
> 
> **Working Format:** 2025-04-12T01:08:13Z
> 
> The ear2earth application was updated to handle both formats. Commit [667dc24](https://github.com/OPEnSLab-OSU/WeatherChimesWeb/commit/667dc24a1ee334fe2cf0272fef1477d899ca50a9)
> 
> This is how SpringBreakChimeTest.EZchime5 looks when loaded post-update. [@udellc](https://github.com/udellc)
> 
> <img alt="Image" width="1466" src="https://private-user-images.githubusercontent.com/27435723/444974151-3c9dd4ee-0cff-4da0-a8eb-927aeac450f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc2MzA4MTEsIm5iZiI6MTc0NzYzMDUxMSwicGF0aCI6Ii8yNzQzNTcyMy80NDQ5NzQxNTEtM2M5ZGQ0ZWUtMGNmZi00ZGEwLWE4ZWItOTI3YWVhYzQ1MGYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTE5VDA0NTUxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzMmVhNTJmY2RkNWVkZjZjODM4MmE5MjUwN2Q4NmViNjZmZTE0MWFlMjdkZjM3OTEzNzE3MjcyYjc5YjAyOWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JmcgFGDg69HU7n1QmatF4A_dQ05yqINoH05fc2-nfzk">
> [@WL-Richards](https://github.com/WL-Richards), You're right that the local format does not require a Z suffix since it indicates Zulu (UTC) time. I've fixed my script. Thanks.

udellc, 2025-05-19:
@sarveshta mentions: ear2earth issue with Invalid date should be fixed now. Give it a try and let me know if there're any further issues." However on testing: . It seems to be displaying timestamps okay now, I'm not getting the undefined thing for the sets I've tested, but I am still getting the jerkey jittery data that looks extremely like 2 datasets interleaved together. Check out screen shot.

udellc, 2025-05-19:
This is on same repo: SpringBreakChimeTest.EZchime5

[image]

WL-Richards, 2025-05-19:
The data looks fine on the MongoDB side

[image]

udellc, 2025-05-20:
Exactly! the dates make no sense on the ear2earth plot!!!

udellc, 2025-06-02:
Thank you Sarvesh! 
Tess will take some of the fine tuning changes from here, but we will call you in if needed!

closed 2025-06-02 by udellc