OPEnSLab-OSU/Loom-V4 #207: SD Debug File Corruption

SorenEmmons, 2025-06-03:
**Describe the bug**
When using Function logging and SD logging debug features, there is a chance it will save corrupted files to the debug file, which seems to have a correlation to the unit freezing. This issue usually occurs within a week of the units being set up and happens to about 1/5 units set up. The corruption can range in severity from only saving a 2-3 corrupted file to 20+ to the debug file. Will note that the issue does seem to also correlate with a unit's watchdog timer tripping, but it seems not to always be the trigger, with some units having the corruption and not having the watchdog timer trip. This issue also seems to happen to both our old and industrial SD cards. 

**Hardware in Use**
Feather m0
Hypnos 3.3
Weatherchimes PCB with 5V boost 
SparkFun LTE board
Sen55 Sensor 
SHT 31 Sensor 

**To Reproduce**
Steps to reproduce the behavior:
1. Using the Sen55Fixv2 
2. Upload the Wisp_Test_Code that includes the debug code
3. Let the unit run for about a week (You may want several units, as there is a  20% chance of this happening seen from my testing) 
5. Check the SD card for corruption, especially if the unit froze 
6. Should see several corrupted file in the debug folder

**Expected behavior**
What should happen is only funcSumaries_log.txt and output.txt files should be present in the debug file 

**Code**
In the Arduino IDE while editing the main.ino, goto 'Edit' -> 'Copy for HTML' and paste the output of that here

```cpp
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Whisp_brd_v0p4_", 40); //change

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MongoDB mqtt(manager, lte);
//A batch is logged every 5 minutes, so 12 per hour (12 * 6 = 72) so mqtt will publish at batch size of 72/ every 6 hours
Loom_BatchSD batchSD(hypnos, 72);

void isrTrigger()
{
  hypnos.wakeup();
}

void setup() {
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Set the LTE board to only powerup when a batch is ready to be sent
  lte.setBatchSD(batchSD);

  // Both power rails should be on when awake
  hypnos.setWakeConfiguration(POWERRAIL_CONFIG::PR_3V_ON_5V_ON);

  // Only the 5V rail should be on during sleep
  hypnos.setSleepConfiguration(POWERRAIL_CONFIG::PR_3V_OFF_5V_ON);

  // Enable the hypnos rails
  hypnos.enable();

  //Time Sync Using LTE 
  hypnos.setNetworkInterface(&lte);

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  hypnos.networkTimeUpdate();

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  // Log the data to the SD
  hypnos.logToSD();

  // Pass in the batchSD to the mqtt obj to check/ publish a batch of data if ready
  mqtt.publish(batchSD);
  //mqtt.publish();

  // Set the interrupt duration for 5 minutes
  hypnos.setInterruptDuration(TimeSpan(0,0,5,0));

  // Reattach the interrupt
  hypnos.reattachRTCInterrupt();
  
  hypnos.networkTimeUpdate();
  // Set the hypnos to sleep, but with power still being supplied to the 5v rail (wait for serial when testing from a computer)
  hypnos.sleep(false);

  //manager.pause(2000);
}

```

**Output**
Some examples of what the debug file looks like

[image]

[image]

[image]

**Additional context**
According to Weatherchimes, they are not experiencing this issue with similar hardware to us, and discussed the issue with Jacob and we think it may be a memory leak due to the oddity of the issue.

SorenEmmons, 2025-06-03:
Ya it seems like it was writing the output file contents as actual file names super weird but this is the only occurrence I found of that specific corruption

WL-Richards, 2025-06-15:
I think the behavior here is that at some point the file structure of the SD card gets corrupted and the data begins being written directly onto the SD card storage, resulting in the contents of the log showing up as files

trenstonr, 2026-05-05:
Ensure the batch transmit method has a call to check if the SD file we're writing to has not been closed, then close it. After that, transmit data. Then reopen a new SD file. Otherwise, the default is to close once a day and reopen the file.