OPEnSLab-OSU/Loom-V4 #208: 4G board sticking on

SorenEmmons, 2025-06-03:
**Describe the bug**
Appears that the 4G board is staying in transmit/connection mode, indicated by the blue light on the 4G board, and what appears in the hologram, I've only seen it twice so far during testing. On a hologram, it claims to have a constant connected status even though no data is ever actually sent, with it claiming connection sessions are open for about an hour when the typical connection should last 4-5 minutes. 

**Hardware in Use**
Feather m0
Hypnos 3.3
Weatherchimes PCB with 5V boost
SparkFun LTE board
Sen55 Sensor
SHT 31 Sensor

**To Reproduce**
Steps to reproduce the behavior:
1. Using the Sen55Fixv2
2. Using Wisp_Test_Code.ino or any other code that has the mqtt.publish(batchSD) function
3. Check to see if the 4G board stays on even after it should closed the connection indicated by the blue light staying or on hologram the unit still has the connected status. 
4. If it doesn't happen, keep disconnecting power to the unit and powering it on until it happens. 

**Expected behavior**
The 4G board should stop trying to establish a connection or transmit and not stay connected 

**Code**

```cpp
#include <Loom_Manager.h>
#include <Hardware/Loom_Hypnos/Loom_Hypnos.h>

#include <Sensors/Loom_Analog/Loom_Analog.h>
#include <Sensors/I2C/Loom_SEN55/Loom_SEN55.h>
#include <Sensors/I2C/Loom_SHT31/Loom_SHT31.h>

//#include <Sensors/Analog/ACS712/Loom_ACS712.h>

#include <Logger.h>
#include <Internet/Connectivity/Loom_LTE/Loom_LTE.h>
#include <Internet/Connectivity/Loom_Wifi/Loom_Wifi.h>
#include <Internet/Logging/Loom_MongoDB/Loom_MongoDB.h>

Manager manager("Whisp_brd_v0p4_", 40); //change

Loom_Hypnos hypnos(manager, HYPNOS_VERSION::V3_3, TIME_ZONE::PST, true);

Loom_Analog analog(manager);

//Main Air Quality, Temperature, and Humidity Sensing
Loom_SEN55 SEN55(manager);
Loom_SHT31 sht(manager);

//Connectivity
Loom_LTE lte(manager, "hologram", "", "");
Loom_MongoDB mqtt(manager, lte);
//A batch is logged every 5 minutes, so 12 per hour (12 * 6 = 72) so mqtt will publish at batch size of 72/ every 6 hours
Loom_BatchSD batchSD(hypnos, 72);

void isrTrigger()
{
  hypnos.wakeup();
}

void setup() {
  ENABLE_SD_LOGGING;
  ENABLE_FUNC_SUMMARIES;

  // Wait 20 seconds for the serial console to open
  manager.beginSerial();

  // Set the LTE board to only powerup when a batch is ready to be sent
  lte.setBatchSD(batchSD);

  // Both power rails should be on when awake
  hypnos.setWakeConfiguration(POWERRAIL_CONFIG::PR_3V_ON_5V_ON);

  // Only the 5V rail should be on during sleep
  hypnos.setSleepConfiguration(POWERRAIL_CONFIG::PR_3V_OFF_5V_ON);

  // Enable the hypnos rails
  hypnos.enable();

  //Time Sync Using LTE 
  hypnos.setNetworkInterface(&lte);

  // Read the MQTT creds file to supply the device with MQTT credentials
  mqtt.loadConfigFromJSON(hypnos.readFile("mqtt_creds.json"));

  // Initialize all in-use modules
  manager.initialize();

  // Register the ISR and attach to the interrupt
  hypnos.registerInterrupt(isrTrigger);

  hypnos.networkTimeUpdate();

}

void loop() {

  // Measure and package the data
  manager.measure();
  manager.package();

  // Print the current JSON packet
  manager.display_data();

  // Log the data to the SD
  hypnos.logToSD();

  // Pass in the batchSD to the mqtt obj to check/ publish a batch of data if ready
  mqtt.publish(batchSD);
  //mqtt.publish();

  // Set the interrupt duration for 5 minutes
  hypnos.setInterruptDuration(TimeSpan(0,0,5,0));

  // Reattach the interrupt
  hypnos.reattachRTCInterrupt();
  
  hypnos.networkTimeUpdate();
  // Set the hypnos to sleep, but with power still being supplied to the 5v rail (wait for serial when testing from a computer)
  hypnos.sleep(false);

  //manager.pause(2000);
}

```

**Output**
Unfortunately, at this time I don't have output at this time but there is a unit that had debugging on at Oak Creak. I need to retrieve what had this issue.

**Additional context**
This isn't really high priority, just making it known this can be an issue, as I can see if the issue is occurring by checking the hologram when the unit has been deployed, or checking if the blue light is staying on, as it seems like it happens or doesn't happen upon start up.

SorenEmmons, 2025-06-03:
You think it may be because of have we keep the 5V rail on constantly?

WL-Richards, 2025-06-04:
That is most certainly a possibility unsure tho

closed 2026-04-15 by trenstonr